digidotcom/xbee-java #28: XBeeDevice open timeout

joychen130, 2015-01-17:
I use the XBee S2 module as zigbee coorinator, it works find. 
But since I setting serial D6 as RTS flow control  pin, I cant open the xbee device and get below exceptions.

[main] INFO com.digi.xbee.api.XBeeDevice - [COM21 - 115200/8/N/1/N] Connection interface open.
[main] ERROR com.digi.xbee.api.XBeeDevice - There was a timeout while executing the requested operation.
com.digi.xbee.api.exceptions.TimeoutException: There was a timeout while executing the requested operation.
	at com.digi.xbee.api.XBeeDevice.enterATCommandMode(XBeeDevice.java:414)
	at com.digi.xbee.api.XBeeDevice.determineOperatingMode(XBeeDevice.java:364)
	at com.digi.xbee.api.XBeeDevice.open(XBeeDevice.java:207)
	at sm.net.xbee.XBeeConn.Open(XBeeConn.java:183)
	at smarterme.Main.main(Main.java:86)
[main] INFO com.digi.xbee.api.XBeeDevice - [COM21 - 115200/8/N/1/N] Connection interface closed.
com.digi.xbee.api.exceptions.InvalidOperatingModeException: Could not determine operating mode.
	at com.digi.xbee.api.XBeeDevice.open(XBeeDevice.java:212)
	at sm.net.xbee.XBeeConn.Open(XBeeConn.java:183)
	at Main.main(Main.java:86)
java.lang.Exception: XBee open failed
	at Main.main(Main.java:88)
[main] INFO com.digi.xbee.api.XBeeDevice - [COM21 - 115200/8/N/1/N] Connection interface closed.

Does any one known how to slove this issue? Thanks.
My library version is 1.0.1, API mode=1

joychen130, 2015-01-17:
I solved it by myself. :)
I forget set "hardware flow control" when the device open :p

closed 2015-02-03 by rubenmoral

etaolla, 2016-03-18:
joychen130, I have the same issue. What do you mean by setting hardware flow control?
On XTCH or a Java function?