digidotcom/xbee-java #29: Get RSSI Value

jetremote, 2015-02-05:
Hello,

Is possible to get RSSI value with this library?

Best regards.

daescalona, 2015-02-06:
Hello,

There is not an specific way to read the RSSI value of a module, but you can do it by reading the 'DB' parameter. The 'DB' parameter stores the RSSI value of the last received packet from any source. This means that, if you want to measure the RSSI value of your local device against a remote device you should perform the following:

* Send any packet to the remote device you want to measure its RSSI value.
* Read the value of the 'DB' parameter of that remote device. This will give you the signal quality of the remote device against your local device.
* Read the value of the 'DB' parameter of your local device. This will give you the signal quality of your local device against the remote device.

You can also use the Range Test feature of the XCTU which basically does the same but in a loop and displaying measures in a chart.

Hope that helps.

jetremote, 2015-02-06:
Thank you very much!

Very useful for me.

Best regards,
--javier

daescalona, 2015-02-09:
Add specific abstraction methods to read last received packet RSSI value.

jetremote, 2015-02-11:
Thank you!

I'll try as soon as I can

rubenmoral, 2016-11-14:
I have created a new story in our backlog to create those abstraction methods.

closed 2016-11-14 by rubenmoral