digidotcom/xbee-java #30: XBee Pro S3B

jetremote, 2015-02-11:
Hello!,

XBeeJavaLibrary is compatible with XBee Pro S3B ?

http://www.digi.com/products/wireless-wired-embedded-solutions/zigbee-rf-modules/point-multipoint-rfmodules/xbee-pro-xsc

I used XBee Pro S2B, works fine to 1 mile (with antenna, grid parabolic 2.4Ghz 24dBi, at 3 meters of high) but I need more range, at least 5 miles. Is it possible to import to Spain this model? I need 30 units, for now.

Best regards

jetremote, 2015-02-17:
Hello!,

Finally I will use this model for long ranges:

http://www.digi.com/products/model?mid=3194

It is valid for Europe.

But is compatible with XBeeJavaLibrary ?

Best regards,
Javier

tatianaleon, 2015-02-20:
Hi!

The XBee PRO 868 is compatible with the XBee Java Library.
The library supports ZigBee, 802.15.4, DigiMesh and Point-to-Multipoint (aka Multipoint proprietary) XBee devices.

To know the protocols supported by every module you can have a look to the XBee Comparison matrix at http://www.digi.com/pdf/chart_xbee_rf_features.pdf

Best Regards

jetremote, 2015-03-15:
Thank you!,

Best regards

jetremote, 2015-05-01:
Hello,

I bought two XBP08-DPSIT-024-ND

But I not find the firmaware for API MODE.

This library works with the firmware 1061 (Single Channel) for XBP08 series?

Regards

wz2b, 2015-05-01:
That's a digi-mesh module ... I tend to think no.

jetremote, 2015-05-01:
Sure?

digi-mesh is S8 Hardware, wich I have is S5 Hardware

Greetings

jetremote, 2015-05-01:
Sorry,

You're referring to  firmware 1061 ?

jetremote, 2015-05-01:
Ok,

I have to use XBee device class.

https://docs.digi.com/display/XBJLIB/Working+with+XBee+classes

I guess this (for XBP08-DPSIT-024 model):

XBee Point-to-multipoint device (DigiPointDevice)

ftp://ftp1.digi.com/support/documentation/xbjlib/javadoc_1.0.0/com/digi/xbee/api/DigiPointDevice.html

jetremote, 2015-05-01:
I changed XBeeDevice by DigiPointDevice and XbeeNetwork by DigiPointNetwork but I get this error:

```may 01, 2015 8:32:57 PM com.digi.xbee.api.XBeeDevice open
INFO: [/dev/ttyUSB0 - 9600/8/N/1/N] Opening the connection interface...
WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.2pre1
	native lib Version = RXTX-2.2pre2
may 01, 2015 8:32:58 PM com.digi.xbee.api.XBeeDevice open
INFO: [/dev/ttyUSB0 - 9600/8/N/1/N] Connection interface open.
may 01, 2015 8:33:03 PM com.digi.xbee.api.XBeeDevice close
INFORMACIÓN: [/dev/ttyUSB0 - 9600/8/N/1/N] Connection interface closed.
Error opening localXBee com.digi.xbee.api.exceptions.InvalidOperatingModeException: Unsupported operating mode: AT mode```

wz2b, 2015-05-01:
Those modules don't have "AT" mode?  I'm not really sure what that means.
It may be when I send some requests for join notifications ....

I don' tknow anything about the non-zigbee modules.  Do they support API
mode 2 (with escaping) ?

jetremote, 2015-05-01:
I think that only works with AT Mode:

http://ftp1.digi.com/support/documentation/html/90001439/com/digi/xbee/api/models/OperatingMode.html

is so?

tatianaleon, 2015-05-01:
Hi jetremote,

The library only supports modules configured in API mode (with or without escapes) and not in transparent (AT) mode.

An XBP08-DPSIT-024 is an XBee-PRO 868. Their operating mode is determined by the AP parameter:

* AP=0: Transparent mode.
* AP=1: API mode.
* AP=2: API escaped mode. 

Transparent mode (AP=0) is the default value. You can use the XCTU (http://www.digi.com/xctu) to configure your radios to work in API (AP=1 or AP=2).

The XBee-PRO 868 reference manual contains more information about the configuration (
http://www.digi.com/support/getasset?fn=90001020&tp=3)

Best Regards

jetremote, 2015-05-02:
Solved!, Thank you!

Now my firmware is 1061 (Newest) - Sincle Channel, API 2

Best regards

closed 2015-05-02 by tatianaleon