digidotcom/xbee-java #31: XBeeJavaLibrary In Android

jetremote, 2015-02-11:
Hello!,

Works XBee (and XBeeJavaLibrary) with OTG Android Devices?

Regards

daescalona, 2015-02-12:
Hello,

We are working onthe Android support. It is still experimental and is not documented, but you can have a look at this branch:

https://github.com/digidotcom/XBeeJavaLibrary/tree/android_support

jetremote, 2015-02-12:
Thank you,

I used Idea (Android Studio), with Gradle http://android-hispano.blogspot.com.es/  (also Eclipse but not for Android). I'll try, I think it works well.

But my Galaxy S4 not works with XBee, I think I have to buy a smartphone or tablet that supports OTG, is it not? Do you use a device with OTG?

Regards,

jetremote, 2015-02-12:
On second thought I will not use a smartphone or tablet because the battery runs out very quickly.

t3db0t, 2015-10-26:
What's the status of the Android Support branch? I need to implement this ASAP; but this branch is 92 commits behind master.  Is it useable in any way? Really the only thing I need to do is use a vendor's proprietary serial port class/methods, can IConnectionInterface be used for this?

daescalona, 2015-10-26:
Hello,

I just merged master on the Android branch. Build is ok and all tests are passing so it is safe to use.

Kind regards.

t3db0t, 2015-10-26:
Wow, awesome! Thank you!

t3db0t, 2015-10-26:
Followup question—is there an example of how to use a separate serial port with the library? Thank you!

daescalona, 2015-10-27:
Hello,

What do you mean with separate serial port? If what you want is to integrate another serial library, no, there is not any example. Best way is to check how it is done with the current one in code. Also, I think there is a pull request with another serial library implementation you could check.

Kind regards.

t3db0t, 2015-11-11:
Hi again—I'm fairly new to Android so I'm still trying to figure out the correct way to do this.  I have a proprietary android tablet with a built-in serial port.  They have their own proprietary serial API (i.e. `GemvaryPortSdk.Start("ttyS0",115200)`). I will have to somehow use that serial port with this library. Is that possible?  Secondly, I assume the builds available on the Releases page would not be useable since they reflect the master branch? Thank you for your time :)

daescalona, 2015-11-13:
Hello,

In order to achieve it you will need to create a class implementing the IConnectionInterface implementing all the required methods using your Serial Port class. Then you will have o instantiate your XBee Device using your newly created Connection class.

Another option is to use the USB connection directly for which we already have support in the Android branch (it uses the Android USB Host interface). With this option you will not need to implement anything new, it works out of the box.

Hope that helps.

Kind regards.

bveina, 2016-10-31:
is there any plan to pull the android branch into the main branch?

rubenmoral, 2016-11-14:
Hi @bveina,

Yes, our plan is to merge and release that functionality very soon.

Best regards.

rubenmoral, 2017-01-13:
Hi everyone,

We've just released a new version of the library, 1.2.0, which includes full support for Android.

Thanks!

closed 2017-01-13 by rubenmoral