digidotcom/xbee-java #32: Maven Repository to digidotcom/XBeeJavaLibrary

jetremote, 2015-04-03:
Hello,

This library not is in Maven Repositories?

http://search.maven.org/#search%7Cga%7C1%7Cxbee

http://mvnrepository.com/search?q=xbee

Regards

wz2b, 2015-04-03:
Hi there,

I released it into Maven Central here:

  http://search.maven.org/#artifactdetails%7Ccom.autofrog.xbee%7Cxbee-api-java%7C0.1.3%7Cjar

so you should be able to get it with:

<dependency>
    <groupId>com.autofrog.xbee</groupId>
    <artifactId>xbee-api-java</artifactId>
    <version>0.1.3</version>
</dependency>

I'm not really sure how things get into jcentral or maven.org though - I thought they'd get mirrored back out.

I'll be putting out some updates soon including a companion that decodes a few ZDO/ZCL messages.  It's far from complete ... but at least a starting point.  Let me know if any of this is useful to you.  I know there are other libraries out there, my goal was to try to be simple and reduce external dependencies.

--C

jetremote, 2015-04-03:
Thank you!, but I use these imports:

import com.digi.xbee.api.RemoteXBeeDevice;
import com.digi.xbee.api.XBeeDevice;
import com.digi.xbee.api.XBeeNetwork;
import com.digi.xbee.api.exceptions.XBeeException;
import com.digi.xbee.api.io.IOLine;
import com.digi.xbee.api.io.IOMode;
import com.digi.xbee.api.io.IOValue;
import com.digi.xbee.api.models.XBee64BitAddress;
import com.digi.xbee.api.utils.ByteUtils;

I added this dependency to my pom:

  	<dependency>
  		<groupId>com.autofrog.xbee</groupId>
  		<artifactId>xbee-api-java</artifactId>
  		<version>0.1.3</version>
  	</dependency>

And I changed the imports by these but cannot be resolved.

import com.autofrog.xbee.api.RemoteXBeeDevice;
import com.autofrog.xbee.api.XBeeDevice;
import com.autofrog.xbee.api.XBeeNetwork;
import com.autofrog.xbee.api.exceptions.XBeeException;
import com.autofrog.xbee.api.io.IOLine;
import com.autofrog.xbee.api.io.IOMode;
import com.autofrog.xbee.api.io.IOValue;
import com.autofrog.xbee.api.models.XBee64BitAddress;
import com.autofrog.xbee.api.utils.ByteUtils;

Regards

jetremote, 2015-04-04:
For the moment I installled the library in my local maven repository so:

~/git/XBeeJavaLibrary$ mvn install:install-file -Dfile=/home/ubuntu/git/XBeeJavaLibrary/target/library/xbjlib-1.0.1.jar -DgroupId=com.digi.xbee.api -DartifactId=xbjlib -Dversion=1.0.1 -Dpackaging=jar
[INFO] Scanning for projects...
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO] 
[INFO] XBee Java Library Project
[INFO] XBee Java Library
[INFO] XBee Java Library Distribution
[INFO] Receive Modem Status Sample
[INFO] Receive Data Sample
[INFO] Receive Data Polling Sample
[INFO] Send Broadcast Data Sample
[INFO] Send Data Asynchronously Sample
[INFO] Send Data Sample
[INFO] Manage Common Parameters Sample
[INFO] Reset Sample
[INFO] Set And Get Parameters Sample
[INFO] IO Sampling Sample
[INFO] Local ADC Sample
[INFO] Local DIO Sample
[INFO] Remote ADC Sample
[INFO] Remote DIO Sample
[INFO] Discover Devices Sample
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building XBee Java Library Project 1.0.1
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-install-plugin:2.3:install-file (default-cli) @ xbjlib-parent ---
[INFO] Installing /home/ubuntu/git/XBeeJavaLibrary/target/library/xbjlib-1.0.1.jar to /home/ubuntu/.m2/repository/com/digi/xbee/api/xbjlib/1.0.1/xbjlib-1.0.1.jar
[INFO] Installing /tmp/mvninstall2651855932408127226.pom to /home/ubuntu/.m2/repository/com/digi/xbee/api/xbjlib/1.0.1/xbjlib-1.0.1.pom
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] XBee Java Library Project ......................... SUCCESS [0.401s]
[INFO] XBee Java Library ................................. SKIPPED
[INFO] XBee Java Library Distribution .................... SKIPPED
[INFO] Receive Modem Status Sample ....................... SKIPPED
[INFO] Receive Data Sample ............................... SKIPPED
[INFO] Receive Data Polling Sample ....................... SKIPPED
[INFO] Send Broadcast Data Sample ........................ SKIPPED
[INFO] Send Data Asynchronously Sample ................... SKIPPED
[INFO] Send Data Sample .................................. SKIPPED
[INFO] Manage Common Parameters Sample ................... SKIPPED
[INFO] Reset Sample ...................................... SKIPPED
[INFO] Set And Get Parameters Sample ..................... SKIPPED
[INFO] IO Sampling Sample ................................ SKIPPED
[INFO] Local ADC Sample .................................. SKIPPED
[INFO] Local DIO Sample .................................. SKIPPED
[INFO] Remote ADC Sample ................................. SKIPPED
[INFO] Remote DIO Sample ................................. SKIPPED
[INFO] Discover Devices Sample ........................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 1.000s
[INFO] Finished at: Sat Apr 04 08:46:53 WEST 2015
[INFO] Final Memory: 6M/102M
[INFO] ------------------------------------------------------------------------

posborne, 2015-04-04:
Hi @jetremote.  I work for Digi but not on the team responsible working on this library.  Having the lib on Maven Central is obviously important, so I'll follow up with them to make sure that happens.  Glad you found a workaround in the short term.  @wz2b. I'm not sure if your stuff is a fork or a different project; we welcome Pull Requests if there is functionality that is missing (though you are not obligated to do so by the license).

wz2b, 2015-04-04:
OK that sounds great.  Thanks for your help ... I know I haven't been super
helpful yet, I am trying to get too much stuff gong at once and am planning
on going back and doing a clean-up run.

The funny thing is that after I actually released the first version of this
I found a bunch of other libraries that overlap with what I'm doing.  So I
still hope I'm not wasting my time here.  My whole goal was to try to be
dependency free (as much as possible) ... and painless but the fact that
you had these issues makes me wonder if I missed the mark.

If you write issues/tickets though I won't ignore them.

jetremote, 2015-04-06:
@wz2b Sorry, it is possible, I do not configured well. Anyway thank you very much.

@posborne  Excellent news, thank you very much  :+1:

posborne, 2015-04-06:
Hi @jetremote, I did a little work this afternoon to try to put something together.  Unfortunately, RxTx can be a bit tricky to work with.  Here's what I have been working with so far (using gradle).  You should be able to make something similar work using maven:

```groovy
apply plugin: 'java'
apply plugin: 'eclipse'

repositories {
    maven { url "http://dl.bintray.com/digidotcom/maven" }
    maven { url "http://ftp1.digi.com/support/m-repo" }
    mavenCentral()
}

dependencies {
    compile group: 'com.digi.xbee', name: 'xbjlib', version: '1.0.1'
}
```

This is based on the 1.0.1 library that has now bee uploaded to https://bintray.com/digidotcom/maven/xbjlib/view.  This is still very much-so a work in progress but would love your feedback to see if this works for you.

posborne, 2015-04-07:
I have tested the following with gradle under linux x86_64.  You should be able to pull in the right rxtx-native library more automatically with a bit more work.  Note that things are now in jcenter which is more likely to already be included in projects:

```groovy
apply plugin: 'java'
apply plugin: 'eclipse'

repositories {
    maven { url "http://ftp1.digi.com/support/m-repo" }
    jcenter()
    mavenCentral()
}

dependencies {
    compile group: 'com.digi.xbee', name: 'xbjlib', version: '1.0.1'
    compile group: 'org.rxtx', name: 'rxtx-native', version: '2.2', classifier: 'linux_x86_64'
}
```

jetremote, 2015-04-09:
Hi Paul,

Nice work!, work fine to me.

Thank you very much!

https://github.com/jetremote/JetRemote/blob/jetremote-pocketsphinx-rpi/pom.xml

Best regards,
-javier

gaetancollaud, 2015-05-24:
Thank you for this. Can you put this information in the main readme so we don't have to go in issues to find this. ;)

Regards

Gaetan

posborne, 2015-05-24:
@tatianaleon, @rubenmoral, @diescalo

To close out this issue, we would like to officially document and set in place a process for releasing xbjl so that it can be installed via maven.  For the previous release, I put the artifact on bintray for download (there is a digi organizaton to which you can be added).  Alternatively, if we place the xbjl jar with the other libraries on the support site, I think that would work as well.  Either way, I think this is a very reasonable (preferred for serious developers) way to depend on the library, so we should support and document it.

tyrel, 2016-04-14:
Hi! I'm starting XBee development and was looking for it in Maven. I see in a post above that @posborne provided a link to a Maven respository at http://ftp1.digi.com/support/m-repo -- however, I get 403 Forbidden when attempting to access this URL.

I do see it on JCenter: https://bintray.com/digidotcom/maven/xbjlib/view

Having used Maven for Java projects at home and work for 5 years, and never having heard of JCenter, I was surprised to see the comment "things are now in jcenter which is more likely to already be included in projects". Everything I've ever needed has come from Maven Central.

Also, the build on JCenter is over a year old now. It looks like you've had releases since then, including one last week.

What is the status of this? Can you provide up-to-date binaries in Maven Central (ideally) or in JCenter?