digidotcom/xbee-java #33: XBee Pro 868 with XBeeJavaLibrary

jetremote, 2015-05-02:
Hello,

I'm testing with this code, attempt to communicate  a Coordinator with a endDevice:

>import com.digi.xbee.api.DigiPointDevice;
import com.digi.xbee.api.DigiPointNetwork;
import com.digi.xbee.api.RemoteDigiPointDevice;
import com.digi.xbee.api.RemoteXBeeDevice;
import com.digi.xbee.api.exceptions.TimeoutException;
import com.digi.xbee.api.exceptions.XBeeException;
import com.digi.xbee.api.io.IOLine;
import com.digi.xbee.api.io.IOValue;
import com.digi.xbee.api.models.XBee64BitAddress;

>	private DigiPointDevice localXBee;
	private DigiPointNetwork xbeeNetwork;
	private RemoteDigiPointDevice selectedNode;
	private RemoteDigiPointDevice node6;

>	// Initialize a local XBee (coordinator) 
		localXBee = new DigiPointDevice("/dev/ttyUSB0", 9600);
			try {
				localXBee.open();
			} catch (XBeeException e) {
				System.out.println("Error opening localXBee " + e);
			}

>		if(localXBee.isOpen()) {
			// Obtain the remote XBee device from the XBee network.
				xbeeNetwork = (DigiPointNetwork) localXBee.getNetwork();
				
>			// Instantiate the remotes XBee devices
				XBee64BitAddress node6Address = new XBee64BitAddress("0013A20040A2C795");
				node6 = new RemoteDigiPointDevice(localXBee, node6Address);
				xbeeNetwork.addRemoteDevice(node6);

I tried with API 1 and 2, but get this error.

>may 02, 2015 1:21:42 PM com.digi.xbee.api.XBeeDevice open
INFORMACIÓN: [/dev/ttyUSB0 - 9600/8/N/1/N] Opening the connection interface...
WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.2pre1
	native lib Version = RXTX-2.2pre2
RXTX Warning:  Removing stale lock file. /var/lock/LCK..ttyUSB0
may 02, 2015 1:21:42 PM com.digi.xbee.api.XBeeDevice open
INFORMACIÓN: [/dev/ttyUSB0 - 9600/8/N/1/N] Connection interface open.
com.digi.xbee.api.exceptions.TimeoutException: There was a timeout while executing the requested operation.
	at com.digi.xbee.api.AbstractXBeeDevice.sendXBeePacket(AbstractXBeeDevice.java:991)
	at com.digi.xbee.api.AbstractXBeeDevice.sendATCommand(AbstractXBeeDevice.java:806)
	at com.digi.xbee.api.AbstractXBeeDevice.sendParameter(AbstractXBeeDevice.java:1985)
	at com.digi.xbee.api.AbstractXBeeDevice.setParameter(AbstractXBeeDevice.java:1908)
	at com.digi.xbee.api.AbstractXBeeDevice.setDIOValue(AbstractXBeeDevice.java:1294)

What am I doing wrong?

Best regards

tatianaleon, 2015-05-02:
Are the local device (connected to ```/dev/ttyUSB0```) and the remote device represented by the ```node``` object in the same network (both must share the same Modem VID (ID) value)? If not the communication won't be possible.

Note the ```xbeeNetwork.addRemoteDevice(node);``` does not join ```node``` to the network, see https://docs.digi.com/display/XBJLIB/Add+and+remove+devices+manually or http://ftp1.digi.com/support/documentation/xbjlib/javadoc/javadoc_1.0.1/com/digi/xbee/api/XBeeNetwork.html#addRemoteDevice-com.digi.xbee.api.RemoteXBeeDevice-

If the remote device has a node identifier (NI) already configured you can use ```xbeeNetwork.discoverDevice("configured_NI");``` (http://ftp1.digi.com/support/documentation/xbjlib/javadoc/javadoc_1.0.1/com/digi/xbee/api/XBeeNetwork.html#discoverDevice-java.lang.String-) to discover that device if it is in the same network.

Or you can discover all the devices in the network using ```xbeeNetwork.startDiscoveryProcess()``` with a listener (see example [DiscoverDevicesSample](https://github.com/digidotcom/XBeeJavaLibrary/blob/master/examples/network/DiscoverDevicesSample/src/com/digi/xbee/api/discoverdevices/MainApp.java)) or waiting to the process to finish (```xbeeNetwork.isDiscoveryRunning()```) and then get the complete list (```xbeeNetwork.getDevices()```).

All the information about discovering devices in the same network here: https://docs.digi.com/display/XBJLIB/Discover+the+network

jetremote, 2015-05-02:
Thank you!

Modem ID it is the same for both modules.

Here is my configuration:

http://www.digi.com/support/forum/47425/could-not-initialize-device

Greetings

jetremote, 2015-05-02:
I tried this code to discover a remote node but after waiting for the time of discovery returns null.

    
    import com.digi.xbee.api.DigiPointDevice;
    import com.digi.xbee.api.DigiPointNetwork;
    import com.digi.xbee.api.RemoteDigiPointDevice;
    import com.digi.xbee.api.exceptions.XBeeException;

    public class Main {

	public static void main(String[] args) throws XBeeException {
		// Instantiate an DigiPointDevice object.
		DigiPointDevice myLocalDevice = new DigiPointDevice("/dev/ttyUSB0", 9600);
		myLocalDevice.open();
		 
		// Get the DigiPoint Network object from the DigiPoint Device.
		DigiPointNetwork network = (DigiPointNetwork) myLocalDevice.getNetwork();
		 
		// Discover the remote device whose node ID is ‘6’.
		RemoteDigiPointDevice discoveredDevice =  (RemoteDigiPointDevice) network.discoverDevice("6");
		
		System.out.print(discoveredDevice.getHardwareVersion());
	}
    }

>INFORMACIÓN: [/dev/ttyUSB0 - 9600/8/N/1/N] Connection interface open.
Exception in thread "main" java.lang.NullPointerException
	at Main.main(Main.java:20)

tatianaleon, 2015-05-04:
Hi jetremote,

By configuring CE=1 (coordinator) in one module and CE=2 (end device) in the other, you are using indirect messaging and polling that is thought to be used with sleeping devices. Basically, this means:

* The coordinator will not send immediately the messages to the end device, it will buffer them till a poll request from the end device arrives. If no poll request is received the message will be discarded after 2.5*SP ms.
* The end device will send a poll request to the messaging coordinator every 100ms when it is awake. This messaging coordinator is established by configuring the end device DH/DL with the SH/SL of the coordinator.

More info in the _Indirect messaging and polling_ topic of the XBee-PRO 868 reference manual (http://ftp1.digi.com/support/documentation/90001020_E.pdf).

So, if you are not going to use sleep devices, it is better to leave CE by default (CE=0) in all of them. Otherwise, you need to configure the DH/DL of your end devices with the SH/SL of the coordinator.

Best Regards

jetremote, 2015-05-04:
Thank you for yours help!

Now all works.

Best regards

closed 2015-05-04 by tatianaleon