digidotcom/xbee-java #36: Node Identification Indicator (0x95) XBeeAPIPacket in the future?

murraytodd, 2015-05-12:
Are there any plans to create an Node Identifier packet in the future? I've got my endpoint nodes sleeping very long periods (several hours with waking time just long enough to send sensor data because they run on very small batteries and depend on small solar cells to recharge) so the standard network discovery doesn't seem applicable.

I've set the endpoints to send their node identification packets upon association, and that helps my server identify them and figure which NI goes with which 16-bit address.

I've got a hack together right now that just pulls apart the payload of the UnknownXBeePacket and figures things out manually, so I'm not stuck. But I thought I would write this as a +1 for having that part of the API represented. (I could try writing it if the authors would welcome the pull request....)

daescalona, 2015-05-13:
Hello Murray,

That package is not yet created in the API, but you are very welcome to fork the repository and perform pull requests with changes and improvements, that is the main goal of making this library open source.

Kind regards.

murraytodd, 2015-05-13:
Cool. If I have the time to do something a bit more clean than my current hack, I'll do so. I just wanted to check that something wasn't already underway, I guess.

rubenmoral, 2016-11-14:
I have created a new story in our backlog for this task.

closed 2016-11-14 by rubenmoral