digidotcom/xbee-java #38: exceptions.TimeoutException: There was a timeout while executing the requested operation.

GiGoShIjAkOvSkI, 2015-05-17:
Hello to everyone,

i have a xbee and i what to use this java library but i have one problem i got a error:

com.digi.xbee.api.exceptions.TimeoutException: There was a timeout while executing the requested operation.
	at com.digi.xbee.api.AbstractXBeeDevice.sendXBeePacket(AbstractXBeeDevice.java:989)
	at com.digi.xbee.api.AbstractXBeeDevice.sendATCommand(AbstractXBeeDevice.java:804)
	at com.digi.xbee.api.AbstractXBeeDevice.sendParameter(AbstractXBeeDevice.java:1983)
	at com.digi.xbee.api.AbstractXBeeDevice.getParameter(AbstractXBeeDevice.java:1925)
	at com.digi.xbee.api.AbstractXBeeDevice.readIOSample(AbstractXBeeDevice.java:1781)
	at com.digi.xbee.api.AbstractXBeeDevice.getDIOValue(AbstractXBeeDevice.java:1334)
	at xbee_discover.XBEE_DISCOVER$UpdateOutputTask.run(XBEE_DISCOVER.java:140)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)

the code:
            ..............................
            ..............................
            setDIOTimer1.scheduleAtFixedRate(new UpdateOutputTask(remote_dev1), 0,5000);
            setDIOTimer2.scheduleAtFixedRate(new UpdateOutputTask(remote_dev2), 5000,10000);
            setDIOTimer3.scheduleAtFixedRate(new UpdateOutputTask(remote_dev3), 10000,20000);
            
            
        } catch (XBeeException e) {
            System.out.println(e);
            myDevice.close();
            System.exit(1);
        }
    }
    
    private static class UpdateOutputTask extends TimerTask {

        private final RemoteXBeeDevice remoteDevice;
        
        public UpdateOutputTask(RemoteXBeeDevice remote) {            
            this.remoteDevice = remote;
        }
        
        @Override
        public void run() {

                
            try {
                if (remoteDevice.getDIOValue(IOLine.DIO3_AD3) == IOValue.HIGH) {
                    remoteDevice.setDIOValue(IOLine.DIO3_AD3, IOValue.LOW);
                } else {
                    remoteDevice.setDIOValue(IOLine.DIO3_AD3, IOValue.HIGH);
                }
            } catch (XBeeException ex) {
                Logger.getLogger(XBEE_DISCOVER.class.getName()).log(Level.SEVERE, null, ex);
            }
                               

        }
    }

Why i get this exception and how to solve?

GiGoShIjAkOvSkI, 2015-05-17:
i debug and view that the exception is throw here 

// Write the packet data.
			writePacket(packet);
			try {
				// Wait for response or timeout.
				synchronized (responseList) {
					try {
						responseList.wait(receiveTimeout);
					} catch (InterruptedException e) {}
				}
				// After the wait check if we received any response, if not throw timeout exception.
				if (responseList.size() < 1)
					throw new TimeoutException();
				// Return the received packet.
				return responseList.get(0);

I set the receiveTimeout to 4000 but i get the exception!Why?

daescalona, 2015-05-18:
Hello,

The timeout exception occurs because you do not receive an answer back from the device. As I saw in the example, you are trying to configure DIO pins of a remote device. Have you verified if there is connection with the remote divice? (fro example performing a device discovery and checking that the device appears in the list).

Also, is the remote device a sleeping device? if so there is a chance the device was sleeping while you were sending the command.

GiGoShIjAkOvSkI, 2015-05-19:
Hello, 

Thank you for your answer. For your first question do I verify the connection with the remote device the answer is yes. I performed device discovery in the beginning of the program and in the list of discovered devices I see my remote device. For your second question, in my network I have 3 devices, 2 routers and one END device and the error I get is from the router, not from the END device (and the router cannot be sleeping). 

I have run the range test ( XCTU), and the results are that from 28 000 240 packets are lost (drop), So my question here is, do I get the exception in the program when a packet is dropped (lost) ?

Thank you in advance

diescalo, 2015-05-22:
When performing a range test within XCTU it considers that a packet is lost when it does not receive the answer for that packet in a period of time. That time can be configured in the Range Test options. 

The XBee Java Library throws a Timeout Exception when the answer of a frame is not received after a period of time. That time can be changed using the following method in an XBeeDeviceobject:

lastSpark, 2016-02-10:
I also receive the same error:
```
com.digi.xbee.api.exceptions.TimeoutException: There was a timeout while executing the requested operation.
        at com.digi.xbee.api.AbstractXBeeDevice.sendXBeePacket(AbstractXBeeDevice.java:1051)
        at com.digi.xbee.api.AbstractXBeeDevice.sendATCommand(AbstractXBeeDevice.java:866)
        at com.digi.xbee.api.AbstractXBeeDevice.sendParameter(AbstractXBeeDevice.java:2053)
        at com.digi.xbee.api.AbstractXBeeDevice.getParameter(AbstractXBeeDevice.java:1992)
        at com.digi.xbee.api.AbstractXBeeDevice.readIOSample(AbstractXBeeDevice.java:1845)
        at com.digi.xbee.api.AbstractXBeeDevice.getDIOValue(AbstractXBeeDevice.java:1398)
        ...
        at java.util.TimerThread.mainLoop(Timer.java:555)
        at java.util.TimerThread.run(Timer.java:505)
```
increasing the default 2sec to 5 sec of `setReceiveTimeout(int receiveTimeout)` does not resolve my issue. Any other suggestion to resolve this?