digidotcom/xbee-java #39: jSSC (Java Simple Serial Communicator)

gaetancollaud, 2015-05-24:
Hi,

Currently, you use RXTX to communicate with the devices. Have you consider using [jSSC](https://github.com/scream3r/java-simple-serial-connector) instead ?

With RXTX you always need to install the native library and link it to your java project. With jSSC the native library is embedded in the java library. It's very easy to use. I was used to RXTX but now I'm only using jSSC and I'm not disappointed. For instance, [Universal GCode  sender](https://github.com/winder/Universal-G-Code-Sender) made the same choice.

Cheers

Gaetan

tatianaleon, 2015-06-03:
Thank you for your suggestion @gaetancollaud, we will have a look to jSSC library.

One of the things that keep us using RxTx library is the Android support.Anyway you are very welcome if you want to add a new serial connector to the XBee Java library using jSSC. To do so follow the steps described at https://github.com/digidotcom/XBeeJavaLibrary/blob/master/CONTRIBUTING.md

Regards

gaetancollaud, 2015-06-04:
I see that you made a very good abstraction of the Serial communicator. You just miss a good way to switch from one implementation to another (like a factory or something). But that will be easy to implement. 

Anyway, I just forked your repo and I will see if I can do something with jSSC. I will keep you in touch.

Regards

gaetancollaud, 2015-06-04:
I come up with something. I'm looking forward for your feedbacks. I'm not really happy with the way we can choose the implementation that we want but I didn't wanted to change your API. If you have a better idea, let me know.

It works well with my project. ;)