digidotcom/xbee-java #41: No such port: /dev/ttyAMA0

jetremote, 2015-06-08:
Hello,

I have this code to attempt to communicate with the local XBee I have connected to a Raspberry Pi but I get this error:

http://pastebin.com/UNZqU2TE

Jun 08, 2015 6:50:57 AM com.digi.xbee.api.XBeeDevice open
INFO: [/dev/ttyAMA0 - 115200/8/N/1/N] Opening the connection interface...
WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.2pre1
	native lib Version = RXTX-2.2pre2
Error opening localXBee com.digi.xbee.api.exceptions.InvalidInterfaceException: No such port: /dev/ttyAMA0

But if I use Screen I can run AT commands.

sudo screen /dev/ttyAMA0 9600

+++
OK
ATSL
ATSH
etc....

What I can try to solve this?

Regards

tatianaleon, 2015-06-08:
Hi @jetremote 

In the code at http://pastebin.com/UNZqU2TE the connection is being opened using a baudrate of 115200:
```java
// Initialize a local XBee
localXBee = new XBeeDevice("/dev/ttyAMA0", 115200);
```
While later it seems you use 9600 in the command line:
```
sudo screen /dev/ttyAMA0 9600
```
You also must take into account the ```/dev/ttyAMA0``` permissions: you are using ```sudo``` to open the serial port with the ```screen``` command, so ensure you have enough permissions to open the port using your Java application.

Regards

jetremote, 2015-06-08:
Sorry for the confusión,

I have tested both baud ratios. But XBee devices is configured as 9600.

From Eclipse I have not sudo permissions, but from Rasbperry Pi terminal I run with sudo permissions and get the same error.

    sudo java -jar PruebaXBeeUART-0.0.1-SNAPSHOT-jar-with-dependencies.jar 

    Jun 08, 2015 10:59:23 PM com.digi.xbee.api.XBeeDevice open
    INFO: [/dev/ttyAMA0 - 9600/8/N/1/N] Opening the connection interface...
    WARNING:  RXTX Version mismatch
    	Jar version = RXTX-2.2pre1
    	native lib Version = RXTX-2.2pre2
    Error opening localXBee com.digi.xbee.api.exceptions.InvalidInterfaceException: No such port:     /dev/ttyAMA0

Thank you and Regards

posborne, 2015-06-08:
Hi @jetremote, I was curious and looked into this a bit.  It appears that RxTx is a bit stupid (for a lack of a better way to put it) and may require you to create a symlink so that the port has a name that RxTx likes (e.g. `/dev/ttyS<n>`).  It must have a bad heuristic for what a serial port is under Linux (the Right Way would probably be to use udev).  On a related note, this probably counts as a point toward something like https://github.com/digidotcom/XBeeJavaLibrary/pull/40 that provides an alternative to RxTx.

http://stackoverflow.com/questions/9717477/java-rxtxcomm-lib-to-connect-to-dev-ttyacm0

You should also be able to access the serial port without sudo if you add your user to the `dialout` group (`sudo usermod -a -G dialout <username>`) and log out and back into your system.

posborne, 2015-06-08:
There appears to be another potential workaround here: https://bugs.launchpad.net/ubuntu/+source/rxtx/+bug/367833

jetremote, 2015-06-09:
Hi @posborne ,

Thank you for the reply,

You're right, Pi is dialout group.

    groups pi
    pi : pi adm dialout cdrom sudo audio video plugdev games users netdev input spi gpio

I normally use with Sparkfun XBee Explorer USB (USB FT231X) connected to the port USB of the Raspberry Pi. I'm designing a pcb module with Kicad to integrate the Raspberry Pi and XBee and communicate via UART instead of USB.

Is it a good idea or is it better to use USB?  (FT231 or FT232 chip)

Sorry, I have other problem, I need urgently 20 XBee Pro 868 for EU, I will need more but I need for this week 20 units. My usual supplier does not stock. You can send 20 units urgently to my supplier in Minnesota?, So they then sent me to Canary Islands (Spain). ( My contact is jetremote dot canarias @ gmail dot com ).

Sorry for my english.

Best regards

jetremote, 2015-06-09:
This works fine to me:

    sudo ln -s /dev/ttyAMA0 /dev/ttyS88

    pi@raspberrypi ~ $ sudo java -jar PruebaXBeeUART-0.0.1-SNAPSHOT-jar-with-dependencies.jar 
    Jun 09, 2015 10:32:54 AM com.digi.xbee.api.XBeeDevice open
    INFO: [/dev/ttyS88 - 9600/8/N/1/N] Opening the connection interface...
    WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.2pre1
	native lib Version = RXTX-2.2pre2
    RXTX Warning:  Removing stale lock file. /var/lock/LCK..ttyS88
    Jun 09, 2015 10:32:55 AM com.digi.xbee.api.XBeeDevice open
    INFO: [/dev/ttyS88 - 9600/8/N/1/N] Connection interface open.
    ZigBee : 29A7 : 30

Thank you very much! @tatianaleon and @posborne 

Then I will use UART serial communication, I think it will work well as USB.

Best regards

abythell, 2015-06-09:
Further to what @posborne wrote, you can tell RXTX about non-standard ports by defining the property:

    gnu.io.rxtx.SerialPorts=/dev/ttyAMA0 

You can do this on the command line when you run Java (`-Dgnu.io.rxtxSerialPorts=/dev/ttyAMA0`) or in your code.  

I like to define the property in the code as it makes the application portable without requiring the end-user to create symlinks.  See [How to Use RXTX on Raspberry Pi or BeagleBone](http://angryelectron.com/rxtx-on-raspbian/) for details.

tatianaleon, 2015-06-09:
Hi @jetremote,

The symbolic link to the port is a little bit tricky. RxTx does not recognize all the possible port names on Linux systems, but it does for "ttyS*" (among others). So adding that link is a way to force that recognition.

But since it is possible to have different names, RxTx offers the possibility to specify them with the property ```gnu.io.rxtx.SerialPorts```.

I would go with the solution provided by @abythell to have ```/dev/ttyAMA0``` port recognized by RxTx. Adding user  "pi" to "dialout" group to have enough permissions as @posborne pointed is also a good idea to avoid using ```sudo```.

posborne, 2015-06-09:
This seems to be resolved. so I will close.  I agree with @tatianaleon and @abythell that the symlink approach is a hack and the using the property is a better solution in general.  Looking at https://www.knopflerfish.org/svn/knopflerfish.org/trunk/osgi/bundles_opt/serial/rxtxcomm/src/gnu/io/RXTXCommDriver.java, it looks like you can specify multiple separated by a colon.

closed 2015-06-09 by posborne

abythell, 2015-06-10:
>it looks like you can specify multiple separated by a colon

Officially it is the file path separator, which is a colon only on some platforms.  I believe Linux uses a colon, and Windows uses a semicolon.  I only mention it because it really tripped me up.

posborne, 2015-06-10:
@abythell, you are correct.  Probably best to use `System.getProperty("path.separator", ":")` for the seperator to match the implementation in RxTx.  Source:

```java
	private void addSpecifiedPorts(String names, int PortType)
	{
		final String pathSep = System.getProperty("path.separator", ":");
		final StringTokenizer tok = new StringTokenizer(names, pathSep);

		if (debug)
			System.out.println("\nRXTXCommDriver:addSpecifiedPorts()");
		while (tok.hasMoreElements())
		{
			String PortName = tok.nextToken();

			if (testRead(PortName, PortType))
				CommPortIdentifier.addPortName(PortName,
					PortType, this);
		}
	}
```

jetremote, 2015-06-10:
Works best, thanks.