digidotcom/xbee-java #42: Stepper Motor Driver Semiconductor to XBee Pro 868

jetremote, 2015-06-14:
Hello,

I use XBee Pro 868 with JavaXBeeLibrary in API Mode 2.

I need to control two stepper motors and a linear actuator.(But possibly need to use any more).

You know some Stepper Motor Driver chip that can be connected directly to the XBee to control these motors?

I've seen some models but not sure that can be used for this scenario.

DRV8825
http://www.ti.com/product/drv8825?keyMatch=drv8825&tisearch=Search-EN-Everything

DRV8842
http://www.ti.com/product/drv8842

Will any of these models serve for what I want to do?

Best regards

faludi, 2015-06-15:
I think the DRV8825 might be the better of the two for your purposes. You'll need to control the I/O pins on the XBee to provide the digital inputs to the driver. If you are new to using these drivers, then you might want to prototype first with one of these, which are well documented and should save you some time designing circuits so you could get right to testing your prototype application:
https://www.sparkfun.com/products/12779
https://www.sparkfun.com/products/12859

jetremote, 2015-06-17:
Thank you faludi

I'll try some of those models.

Best regards

closed 2015-06-22 by tatianaleon