digidotcom/xbee-java #43: Get Device by 64-bit Address

jetremote, 2015-06-21:
Hello!,

I need get a device by 64-bit address but I get null because my Node ID is null, but I do not understand why compare 64-bit address with NodeID in XBeeNetwork.clas (line 384):

    	// Look in the 64-bit map.
    		for (RemoteXBeeDevice remote : remotesBy64BitAddr.values()) {
    			if (remote.getNodeID().equals(id))
    				return remote;
    		}

Should not it be something like?

    	// Look in the 64-bit map.
    		for (RemoteXBeeDevice remote : remotesBy64BitAddr.values()) {
    			if (remote.getXbee64BitAddress().equals(id))
    				return remote;
    		}

But I think I'll have to put the 64-bit address in the NodeID, is it so?

Sorry me english.

Best regards.

rubenmoral, 2015-06-22:
Hi @jetremote,

The `XBeeNetwork` class provides 3 methods to get a single device
* By node identifier: `getDevice(String)`
* By 64-bit address: `getDevice(XBee64BitAddress)`
* By 16-bit address: `getDevice(XBee16BitAddress)`

You could use the second one if the node identifier of your remote is empty.

And regarding your question, the code is correct because it belongs to the `getDevice(String)` method. It compares the given node identifier with the node identifier of each remote device in the list.

Thanks for your question!
Best regards.

jetremote, 2015-06-22:
Thank you Ruben!

I use getDevice(XBee64BitAddress) but I get a null exception:

    Caused by: java.lang.NullPointerException
    	at com.digi.xbee.api.XBeeNetwork.getDevice(XBeeNetwork.java:384)
    	at com.jetremote.Main$2.run(Main.java:168)
    	at com.jetremote.Main.main(Main.java:176)
    

I parsed xml file with xbee serials (high and low):

    <?xml version="1.0" encoding="UTF-8"?>
<modules>
    <module id="1">
        <serialhigh>0013A200</serialhigh>
        <seriallow>40A29285</seriallow>
     </module>
    <module id="2">
        <serialhigh>0013A200</serialhigh>
        <seriallow>40A29283</seriallow>
    </module>
   <module id="3">
        <serialhigh>0013A200</serialhigh>
        <seriallow>40A2927E</seriallow>
    </module>
</modules>

This is the code for instantiate the remotes Xbee devices:
    for (int i = 0; i < serials.size(); i++){
			XBee64BitAddress nodeAddress = new XBee64BitAddress(serials.get(i).getSerialHigh()+serials.get(i).getSerialLow());
			digiPointNetwork.addRemoteDevice(new RemoteDigiPointDevice(localXBee, nodeAddress));
		}

When I try to change the value of the pin I get the exception:

    digiPointNetwork.getDevice(address).setDIOValue(IOLine.DIO0_AD0, IOValue.HIGH);

What am I doing wrong?

Best regards

jetremote, 2015-06-22:
I Debug and in this line:

    if (remote.getNodeID().equals(id))

'remote' value is -->  0013A20040A29285 - null

'id' value is -->  0013A20040A2927E

In digiPointNetork I have these devices:
```{0013A20040A29285=0013A20040A29285 - null, 0013A20040A29283=0013A20040A29283 - null, 0013A20040A2927E=0013A20040A2927E - null}```

jetremote, 2015-06-22:
Sorry, this is the java class

Regards

rubenmoral, 2015-06-22:
Hi @jetremote,

I'm doing some tests with DigiPoint devices and all seems to work fine.

Anyway, the exception is thrown in the `getDevice(String)` method (line 384), not in the `getDevice(XBee64BitAddress)` method. This makes me think that the `address` variable could be a String and not an `XBee64BitAddress` object. Since the remote modules have been added to the network manually, their node identifier is null, so that can be the cause of the NullPointerException.

Could you please check the type of `address`?

jetremote, 2015-06-22:
Yes, address is a String

jetremote, 2015-06-22:
Now works fine, Thank you Ruben!

```private static RemoteXBeeDevice getRemoteXbeeDevice(String selection) {
		XBee64BitAddress address = null;
		// 	Get XBeeAddress to selected key
		for(int i = 0; i < serials.size();i++){
			if(serials.get(i).getId().equals(selection)){
				address = new XBee64BitAddress(serials.get(i).getSerialHigh()+serials.get(i).getSerialLow());
			}
		}
		return digiPointNetwork.getDevice(address);
	}```

Kind regards!

rubenmoral, 2015-06-22:
Perfect, you're welcome!

closed 2015-06-22 by rubenmoral