digidotcom/xbee-java #44: There was a timeout while executing the requested operation.

jetremote, 2015-06-26:
Hello,

I'm having this problem when you try to change the value of a remote pin.

https://github.com/digidotcom/XBeeJavaLibrary/issues/27

I had done tests with the same code and worked well but now this error is appearing.

I have all XBee configured as API 2,   AP = 2

Best regards

tatianaleon, 2015-06-27:
Hi @jetremote 

The latest release of the library (1.1.0) already includes a fix for this issue. Please use this version, here you have a link:

http://github.com/digidotcom/XBeeJavaLibrary/releases/tag/v1.1.0

Best Regards

jetremote, 2015-06-27:
Thank you @tatianaleon 

I used maven version

  <repositories>
		<repository>
			<id>ftp1.digi.com</id>
			<name>XBee Repository</name>
			<url>http://ftp1.digi.com/support/m-repo</url>
			<snapshots>
				<enabled>true</enabled>
				<updatePolicy>always</updatePolicy>
			</snapshots>
		</repository>
    </repositories>
.
Best regards

jetremote, 2015-06-27:
I tried this version but I have the same problem.

http://github.com/digidotcom/XBeeJavaLibrary/releases/tag/v1.1.0

I also am having trouble discovering the device remote from XCTU. Sometimes it is uncovered and not others.

I have connected the serial low 40A29226 in XCTU for Linux and a raspberry pi to 279.It recognizes that well, but I have another, 27B, connected to the pcb jetremote and that's the problem I'm having.

The output is 3.33V without connecting the xbee, when I connect the voltage changes to xbee 3.10V, it is normal that the XBee Pro 868 has an impedance of 0.23V?

Regards,

[image]

[image]

jetremote, 2015-06-27:
Hello,

I changed the 27B by other and the other works smoothly. Then I went to use the 27B and also worked well. Now they are communicating all do not know why sometimes happens that mistake.

Best regards

jetremote, 2015-06-28:
Sorry, but I think the mistake is not the Java library.

I'm using Linux and XCTU three remote devices, but can only find 2. Sometimes 3.

I also tried sending frames from the console mode and it works but only for the discovered devices.

I do not understand why sometimes do not find some devices.

All four devices are routers in API 2. And all remote are connected to the same voltage.

Best regards

tatianaleon, 2015-07-13:
Hi @jetremote,

When you say "can only find 2. Sometimes 3", is always the same device the one that sometimes cannot be remotely discovered?
Are the rest always remotely discovered even if you attached a different one to your PC?
Do all your modules have the same configuration?

Best Regards,
Tatiana

jetremote, 2015-07-16:
Hello,

All modules was the same configuration.

Now works with API 1.

Best regards

igoralves98, 2015-11-07:
I had got a lot of timeout errors in the beginning. Setting the ReceiveTimeout to 1000ms solved the problem.

rubenmoral, 2016-11-14:
This seems to be an issue related to the XBee modules and their configuration, not to the XBee Java Library.

closed 2016-11-14 by rubenmoral