digidotcom/xbee-java #45: New version in Maven

jetremote, 2015-06-28:
Hello,

Could you update Maven to the version 1.1.0?

Best regards

posborne, 2015-07-14:
Hi @jetremote, I'll try to get a new version published in the next couple of days.  Sorry about the delay.  Longer term, we'll try to get something in place so it happens in a more timely fashion.

jetremote, 2015-07-16:
Thank you,

Best regards

jon-whit, 2015-07-23:
I'd like to bump this request. It would be nice if Digi would get this project loaded into the [Maven central repository][1]. I'm just starting to use this library on a project of mine, and I'd like to use Maven for my dependency management.

[1]: http://search.maven.org/

rhwood, 2016-10-16:
Will this library be available from maven?

rubenmoral, 2016-11-14:
We'll try to upload the library to Maven Central soon, I have created a task for this. We'll keep you updated on our progress in #32.

closed 2016-11-14 by rubenmoral