digidotcom/xbee-java #46: XBeeJavaLibrary with Xtend-PKG modem

RodrigoMateus, 2015-07-14:
I'm using XBeeJavaLibrary with 2 Xtend-PKG modems. This is possible because the Xtend modems accept frames type "XBee DigiMesh" (Firmware 8062). I successfully tested the examples SendBroadcastDataSample, SendDataSample, ReceiveDataSample and ReceiveModemStatusSample.

The only change I made in the library code was change the "XBeeProtocol" class. In "determineProtocol" method, where is "return XTEND_DM" I switched to "return DIGI_MESH".

I would like to hear from you if I will find any limitations later. Do you recommend other changes?

daescalona, 2015-07-15:
Hello Rodrigo,

Thank you for your feedback. As you correctly supossed, firmware 8XXX of XTend devices correspond to Digi Mesh protocol. This makes it possible for a XTend device to behave like a Digi Mesh device.

The reason to have a "XTEND_DM" protocol in the API would let us differentiate in a future the type of hardware Digi Mesh is running on. At the moment the implementation of the "XTEND_DM" protocol through the API is not complete, and you won't be able to do any specific DigiMesh actions. The change you made however let's you skip all these limitations, and you should not see any problem with the current API version. 

Until the "XTEND_DM" protocol is fully integrated in tthe API you can continue with that change.

Kind regards.

PS: I will maintain this issue open until "XTEND_DM" protocol integration is made.

RodrigoMateus, 2015-07-15:
Good news! XBeeJavaLibrary will greatly facilitate my work.

Thank you!