digidotcom/xbee-java #48: Send large file

RodrigoMateus, 2015-09-07:
Hello,

I need to send an image (100 KB) once a day. What is the best strategy to send a large file? Could you provide an example?

Best regards.

rubenmoral, 2016-11-14:
Hi @RodrigoMateus,

I'm not sure if this is a common use case for XBees. Anyway, a possible solution would be to encode the image to base64, split the resulting string, and send the chunks using one of the `sendData()` methods. Then, in the receiver module, you should combine the chunks to get the original string and finally decode the image.

Best regards.

closed 2016-11-14 by rubenmoral

RodrigoMateus, 2016-11-18:
Hi, Ruben,

I implemented a very similar solution. The only difference is that I use ZIP compression of the String, in addition to reducing the size of the message, I can check the integrity of the file on the target. It worked very well.

Best regards.

brunoaduarte, 2017-01-19:
Hi @RodrigoMateus , i need to provide a very similar functionality to my system, but instead of 100 Kb i need to transfer 30 MB images. Can you share the solution you created ? Thanks

RodrigoMateus, 2017-01-21:
Hi Bruno,

To send a large file, I serialize it to byte [] and divide into 250 byte parts (for XTend radios). Then I send each part using the method "DigiMeshDevice.sendExplicitData(remoteXBeeDevice, sourceEndpoint, destEndpoint, clusterID, profileID, date)" where:

    DestEndpoint = defines the message id.
    ClusterID = defines the number (sequence) of the part.
    Data = byte [] part.

As messages are received, I will copy each part into a byte []. I use the above parameters to join the parts in the correct way. At the end, I deserialize this byte [] to get the original file.

There are several ways to implement this, depends on the number of radios, whether the messages are sent in sequence, whether there is error handling, etc., this is just a general idea.

I hope I have helped.

brunoaduarte, 2017-01-23:
Nice @RodrigoMateus , thanks !

Just one question, how do you get the "explicits" ClusterID and DestEndPoint values when receiving the packet with "DataReceiveListener" ? 

```
public class DataReceiveListener implements IDataReceiveListener {

	public DataReceiveListener() {
	}

	@Override
	public void dataReceived(XBeeMessage xbeeMessage) {
        }
```

XBeeMessage does not provide any methods to get these values.

brunoaduarte, 2017-01-23:
Thank you @RodrigoMateus !

I'm not able to receive the packets using the "ExplicitDataReceiveListener", did you have any problems using it ?

PS: the regular "DataReceiveListener" works fine. But i need the "ExplicitDataReceiveListener" so i can read ClusterID values.

```
    public static void main(String[] args) {

        DigiMeshDevice xbee = new DigiMeshDevice("COM1", 115200);
		try {
			xbee.open();
			xbee.addExplicitDataListener(new ExplicitDataReceiveListener());
//			xbee.addDataListener(new DataReceiveListener());

		} catch (XBeeException e) {
			e.printStackTrace();
			System.exit(1);
		}
    }
```

```
public class ExplicitDataReceiveListener implements IExplicitDataReceiveListener {
	
	@Override
	public void explicitDataReceived(ExplicitXBeeMessage xbeeMessage) {

		System.out.format("From %s >> %s | %s%n", xbeeMessage.getDevice().get64BitAddress(),
				HexUtils.prettyHexString(HexUtils.byteArrayToHexString(xbeeMessage.getData())),
				new String(xbeeMessage.getData()));

        }
}
```