digidotcom/xbee-java #51: RXTX JNI/.so segfaults on Ubuntu 14

T-R0D, 2015-09-16:
Hi, I wrote myself an application using the XBJL on Mac and it works great. however, I need it to run on a Linux machine, specifically Ubuntu 14.04.

However, no matter what combination of RXTX/JavaComm jars and librxtxSerial.so objects (for x86_64) I use, the best result I get is that there is a seg-fault. I have searched around the internet for a couple days now and have been unable to solve the issue.

I know Digi doesn't develop librxtx, but I was wondering if anyone else had run into this (and solved it)?

My output from the MyFirstXBeeApp.java program is as follows:

WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.2pre1
	native lib Version = RXTX-2.2-20081207 Cloudhopper Build rxtx.cloudhopper.net
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007fa1012465da, pid=10699, tid=140330066917120
#
# JRE version: Java(TM) SE Runtime Environment (8.0_60-b27) (build 1.8.0_60-b27)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.60-b23 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [librxtxSerial.so+0x75da]  Java_gnu_io_RXTXPort_nativeDrain+0xea
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/framework/Desktop/MyFirstXBeeApp/hs_err_pid10699.log
[thread 140329484539648 also had an error]
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

changhe3, 2016-04-17:
Same issue here. I switched to RXTX 2.1.7 and still got a segfault.

T-R0D, 2016-07-05:
I know it's been a while, but I did eventually get this to work. It appears what happened was that the project needs to maintain only 1 version of the RXTX binary. I needed my project to work on multiple platforms, so when I kept all of the binaries in the same location, the JVM didn't always select the correct one.

Probably not a perfect solution, but something to watch out for.

closed 2016-07-05 by T-R0D