digidotcom/xbee-java #53: java.lang.NullPointerException: 64-bit address cannot be unknown.

cumpeo, 2016-01-29:
Hello!!
I have this problem: 

`Exception in thread "Thread-2" java.lang.NullPointerException: 64-bit address cannot be unknown.
        at com.digi.xbee.api.XBeeNetwork.getDevice(XBeeNetwork.java:415)
        at com.digi.xbee.api.connection.DataReader.getRemoteXBeeDeviceFromPacket(DataReader.java:588)
        at com.digi.xbee.api.connection.DataReader.packetReceived(DataReader.java:444)
        at com.digi.xbee.api.connection.DataReader.run(DataReader.java:383)`

I used ReceiveDataSample for recive data from arduino with xbee.
I only save data and adreess of xbee.

rubenmoral, 2016-11-14:
Hi @cumpeo,

Could you please give us more details about your issue? What XBee modules and firmware are you using?

Thanks.

lihanyu23, 2017-01-10:
Hi,

I've also encountered similar issue but in the latest release v1.2.0.
The exception thrown has been changed to IllegalArgumentException, but it always crashes the input parsing thread.

This is caused by a frame received having an unknown 64-bit address (FF FF FF FF FF FF FF FF).
The frame is sent by an XB24C family end-device module, probably with firmware version 4023 (2013/2). I don't know yet why this kind of device broadcasts with address masked.

```
D/DataReader( 8464): [/dev/bus/usb/001/004 - 9600/8/N/1/N] Packet received:
D/DataReader( 8464): {}Packet: 7E001590FFFFFFFFFFFFFFFF894501F25020145607100003C2
D/DataReader( 8464): Start delimiter: 7E
D/DataReader( 8464): Length: 00 15 (21)
D/DataReader( 8464): Frame type: 90 (Receive Packet)
D/DataReader( 8464): 64-bit source address: FF FF FF FF FF FF FF FF
D/DataReader( 8464): 16-bit source address: 89 45
D/DataReader( 8464): Receive options: 01
D/DataReader( 8464): RF data: F2 50 20 14 56 07 10 00 03
D/DataReader( 8464): Checksum: C2
D/UsbDeviceConnectionJNI( 8464): close
W/dalvikvm( 8464): threadid=15: thread exiting with uncaught exception (group=0x40fad930)
E/AndroidRuntime( 8464): FATAL EXCEPTION: Thread-227
E/AndroidRuntime( 8464): java.lang.IllegalArgumentException: 64-bit address cannot be unknown.
E/AndroidRuntime( 8464):        at com.digi.xbee.api.XBeeNetwork.getDevice(XBeeNetwork.java:415)
E/AndroidRuntime( 8464):        at com.digi.xbee.api.connection.DataReader.getRemoteXBeeDeviceFromPacket(DataReader.java:698)
E/AndroidRuntime( 8464):        at com.digi.xbee.api.connection.DataReader.packetReceived(DataReader.java:572)
E/AndroidRuntime( 8464):        at com.digi.xbee.api.connection.DataReader.run(DataReader.java:511)
I/Process ( 8464): Sending signal. PID: 8464 SIG: 9
```

The question is why receiving packet containing unknown 64-bit address must cause the application to crash. Is there a workaround or possibly fix.

rubenmoral, 2017-01-10:
Hi @lihanyu23,

Thanks for your comment. We are investigating the issue and why you are receiving an unknown 64-bit address.

Could you please tell us what packet are you sending from the end-device module?

Thanks!

commit e7b006bc (2017-01-10): fix #53: check that 64-bit address is not unknown before obtaining the
remote XBee device from the network

https://github.com/digidotcom/XBeeJavaLibrary/issues/53
closed 2017-01-10 by rubenmoral