digidotcom/xbee-java #57: Could not determine operating mode @ XBee-PRO 900HP module

lucadavoli, 2016-05-23:
The connection with a [XBee-PRO 900HP](http://www.digi.com/products/xbee-rf-solutions/modules/xbee-pro-900hp) module is not possible with XBeeJavaLib v.1.1.1, since the Java application return the following exception:
com.digi.xbee.api.exceptions.InvalidOperatingModeException: Could not determine operating mode.
	at com.digi.xbee.api.XBeeDevice.open(XBeeDevice.java:218)

The COM port selected into the Java code is right.
A test on Linux platform has been also done, but nothing work.
The only way is to open XCTU and let him to discover the module, and then the module will be available and it works, but if the module is powered off and the powered on, the issue still remain.

How is possible to solve this issue?

daescalona, 2016-05-23:
Hello,

To ensure the problem is not due to the module configuration, could you try this please?
Using XCTU:
- Discover the device.
- Save your current configuration profile so you can restore it in the future.
- Configure the module with the default settings (click on "Defaults" button and then click on "Write" button).
- Change the "AP" setting to "API Mode Without Escapes [1]" (The Java API only works with API mode)
- Write the settings.

Then try to launch your Java application again and check if it fails.

Also, which firmware version are you using?

Thanks.

lucadavoli, 2016-05-23:
Hello David,

i already tried to change the AP setting from Transparent Mode to API1 and API2, but if the Xbee module has not been recognized by XCTU tool before the execution of Java application, nothing happens.

Currently i updated the XBee firmware from 8071 to 8074 (the latest available).
This issue can be due to firmware version?

Thanks.

daescalona, 2016-05-23:
Hello,

I just tried with the "Manage Common parameters Sample" using firmware 8074 and it works perfectly. I used a module with all settings set to their default values except the AP setting, which I set to 1. Could you try this setup please?

lucadavoli, 2016-05-23:
Hello,

i tried also this example, both nothing work.

I attached you the configuration used on my module:
`<?xml version="1.0" encoding="UTF-8"?>

<data>
  <profile>
    <description_file>XBP9B_8074.xml</description_file>
    <settings>
      <setting command="CM">00FFFFFFFFFFF7FFFF</setting>
      <setting command="HP">0</setting>
      <setting command="ID">7FFF</setting>
      <setting command="MT">3</setting>
      <setting command="PL">4</setting>
      <setting command="RR">A</setting>
      <setting command="CE">0</setting>
      <setting command="BH">0</setting>
      <setting command="NH">7</setting>
      <setting command="MR">1</setting>
      <setting command="NN">3</setting>
      <setting command="DH">0</setting>
      <setting command="DL">FFFF</setting>
      <setting command="TO">C0</setting>
      <setting command="NI">0x20</setting>
      <setting command="NT">82</setting>
      <setting command="NO">0</setting>
      <setting command="CI">11</setting>
      <setting command="EE">0</setting>
      <setting command="KY"></setting>
      <setting command="BD">3</setting>
      <setting command="NB">0</setting>
      <setting command="SB">0</setting>
      <setting command="RO">3</setting>
      <setting command="FT">13F</setting>
      <setting command="AP">1</setting>
      <setting command="AO">0</setting>
      <setting command="D0">1</setting>
      <setting command="D1">0</setting>
      <setting command="D2">0</setting>
      <setting command="D3">0</setting>
      <setting command="D4">0</setting>
      <setting command="D5">1</setting>
      <setting command="D6">0</setting>
      <setting command="D7">1</setting>
      <setting command="D8">1</setting>
      <setting command="D9">1</setting>
      <setting command="P0">1</setting>
      <setting command="P1">0</setting>
      <setting command="P2">0</setting>
      <setting command="P3">1</setting>
      <setting command="P4">1</setting>
      <setting command="PD">7FFF</setting>
      <setting command="PR">7FFF</setting>
      <setting command="M0">0</setting>
      <setting command="M1">0</setting>
      <setting command="LT">0</setting>
      <setting command="RP">28</setting>
      <setting command="AV">1</setting>
      <setting command="IC">0</setting>
      <setting command="IF">1</setting>
      <setting command="IR">0</setting>
      <setting command="SM">0</setting>
      <setting command="SO">2</setting>
      <setting command="SN">1</setting>
      <setting command="SP">12C</setting>
      <setting command="ST">BB8</setting>
      <setting command="WH">0</setting>
      <setting command="CC">2B</setting>
      <setting command="CT">64</setting>
      <setting command="GT">3E8</setting>
      <setting command="DD">B0000</setting>
    </settings>
  </profile>
</data>
`

Thanks.

daescalona, 2016-05-23:
Hello,

Could you export the profile to an xml file and then attach the xml file directly? (drag and drop here). To export the configuration file, just  click the profile button and select "Save configuration profile" from the drop down menu.

Thanks.

lucadavoli, 2016-05-23:
Hello,

in attachment you can find the XML version of the configuration file.

Thanks.
Luca

lucadavoli, 2016-05-23:
Hello David,

unfortunately GitHub does not let me to upload the XML version of the file.
If you change the file extension from TXT to XML the file become an XML one and it corresponds to the configuration file released to me by XCTU tool.

Thanks again.

daescalona, 2016-05-23:
I just loaded the profile to one of our modules and the "Manage Common parameters Sample" works without problems. I have disconnected the module, connected it again and launched the example to try to reproduce your scenario, but the behavior is the expected one. For this test I am using a Digi XBIB Board.

The problem must be elsewhere, it could be an issue with the hardware you are using to connect the module to the computer. Which board are you using?

Thanks.

lucadavoli, 2016-05-23:
Hello David,

i'm using one of the modules included in a [XBee-PRO 900HP kit](http://www.digi.com/products/models/xkp9-dmb0) in conjuction with the Java library.
I tested both the source library and the already-packed JAR file version, but both does not work.
The only working scenario remains when i let XCTU recognising the module and then running the Java application.

Thanks.

lucadavoli, 2016-05-23:
Hello,

the same Java code works with another Digi product type, [XBee 868LP kit](http://www.digi.com/products/models/xk8-dms-0).
I tried to change also the USB cable connecting the XBee module to my PC, but the result did not change.
If i power on the module, open XCTU and let him to recognize the Xbee module, then power off the module and finally power on it (connecting and disconnecting the USB cable to my PC), the Java application does not work.
The only way is to maintain the XBee module powered on and previously recognized by XCTU tool, otherwise the Java application is unable to recognize the operating mode (that, with XCTU, has been setted to API1).

Thanks again David.

daescalona, 2016-05-23:
That is weird :(

I am afraid I am not able to reproduce your scenario here and can't give much more support. You could try to configure the serial connection to use Hardware Flow Control (anyway I assume you are working at 9600, and HW flow control does not make much sense at that low speed):

XBeeDevice myDevice = new XBeeDevice(PORT, BAUD_RATE, 8, 1, 0, 3);

If the problem persist you should contact with our support team and check if they can help you. I am sorry I could not be of any help.

Tech.Support@digi.com

We will continue investigating the issue here,

Regards.

lucadavoli, 2016-05-24:
Hello David,

thanks a lot for all your help.
Unfortunately i had already contacted the thechnical support team at the mail address that you provide to me, but they replied me that i need to open a thread here, on GitHub, since the problem that i have is related to Java library and not to Xbee hardaware itself.

Thanks again.
Regards.

daescalona, 2016-05-26:
Hello,

We have been looking at this issue and the problem could be that your module is a programmable device. Could you confirm this? Look at the sticker of the back for the device model. The device model of a NO programmable device looks like this: XBP9B-XCWT-001

lucadavoli, 2016-05-26:
Hello David,

i think that my Xbee modules are all pragrammable, since in XCTU i need to check the option "The radio module is programmable".
The Xbee modules have the current text on their rear labels:
- XBP9B-DMSTB002-revE (Model: XBEE PROS3B)
- XBP9B-DMUTB002-revE (Model: XBEE PROS3B)
- XBP9B-DMWTB002-revE (Model: XBEE PROS3B)

Thanks again.

daescalona, 2016-05-26:
Hello Luca,

That explains a lot :) The problem with the programmable XBees is that once the device is powered up, the boot process executes the application that is flashed (bootloader if none). At this point all the commands sent using the serial connection are received by the programmable chip instead of the radio one. What XCTU does with the checkbox is to put the module in bypass mode, thus all the commands received through the serial connection are forwarded to the radio chip.

In order to automatically put the module on bypass mode after device startup you will need to flash the "serial bypass" application to each of the programmable modules.

lucadavoli, 2016-05-26:
Hello David,

thanks a lot! This "serial bypass" application will be installed on the Xbee module with which application? With XCTU?
And where "serial bypass" application is located?

Thanks again for your patience!

daescalona, 2016-05-26:
Hello,

The getting started process of the programmable kit you have bought [XBee-PRO 900HP kit](http://www.digi.com/products/models/xkp9-dmb0) should guide you on how to install the required software (CodeWarrior + Digi Extensions), setup your hardware using the included debugger and then build and flash your first application. The "serial bypass" is a sample application included in the kit than can be flashed following the same procedure.

As a side note, if you do not plan to use the programmable capabilities and features of your XBee modules and your intention is to use only the XBee radio, my recomendation is to use standard XBee modules instead of programmable ones. The programmable XBee devices are powerful and can host custom applications, but they are also more complicated to work with than standard modules.

lucadavoli, 2016-05-26:
Hello David,

i'd really thank you for this solution.
Tomorrow i'd test it, but i think that it can works!

If it works, i close this issue thread.

Thanks again.

closed 2016-06-09 by lucadavoli

StetroF, 2023-09-08:
@lucadavoli 
Hello,
I have same problem with Xbee-python library & XB24C product. do you have a way to solve your problem? Can you share your experience？
Thanks a lot!