digidotcom/xbee-java #60: java.lang.IllegalArgumentException: RSSI value must be between 0 and 100.

SkyMoCo, 2016-09-28:
Every couple hours of running I get this error and I am not sure where it is happening as I am not creating a RX16Packet.   Need help to debug it.

Exception in thread "Thread-4" java.lang.IllegalArgumentException: RSSI value must be between 0 and 100.
        at com.digi.xbee.api.packet.raw.RX16Packet.<init>(RX16Packet.java:131)
        at com.digi.xbee.api.packet.raw.RX16Packet.createPacket(RX16Packet.java:104)
        at com.digi.xbee.api.packet.XBeePacketParser.parsePayload(XBeePacketParser.java:299)
        at com.digi.xbee.api.packet.XBeePacketParser.parsePacket(XBeePacketParser.java:194)
        at com.digi.xbee.api.connection.DataReader.run(DataReader.java:409)

SkyMoCo, 2016-10-04:
It seems to happen if I doing multiple sends of data at a rapid rate at 9600 baud.

rubenmoral, 2016-11-14:
Hi @SkyMoCo,

Could you please give us more context about what you are doing exactly? What XBee modules and firmware are you using? It would really help us to figure out what's going on and fix the issue.

Thanks.