digidotcom/xbee-java #97: Error compiling Maven project with XBeeJavaLibrary "cannot access android.content.Context"

brunoaduarte, 2017-01-27:
I'm having trouble trying to compile a maven project that uses the XBeeJavaLibrary.

> mvn clean compile assembly:single

```
[INFO] Scanning for projects...
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building daemon 0.1.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ daemon ---
[INFO] Deleting /daemon/target
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ daemon ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] skip non existing resourceDirectory /daemon/src/main/resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.3:compile (default-compile) @ daemon ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 24 source files to /daemon/target/classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] /daemon/src/main/java/io/daemon/App.java:[51,31] cannot access android.content.Context class file for android.content.Context not found
[INFO] 1 error
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 7.859 s
[INFO] Finished at: 2017-01-27T15:56:25-02:00
[INFO] Final Memory: 19M/123M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.3:compile (default-compile) on project daemon: Compilation failure
[ERROR] /daemon/src/main/java/io/daemon/App.java:[51,31] cannot access android.content.Context
[ERROR] class file for android.content.Context not found
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
```

I checked the 

> XBeeDevice.class

 file in the library and there's
`import android.content.Context;`

But i don't use Android here.

Anyway to bypass this ?

diescalo, 2017-02-03:
Hello,

Latest version of the library includes support for Android devices and there are new constructors for the XBeeDevice classes that need an Android context. 

How do you define the dependency with the XBee Java Library in your Maven project? The library is now stored in Maven Central, so it should resolve all the dependencies automatically (including the Android one).

If the dependency is configured with the local xbee-java-library-1.2.0.jar file, you may need to add also a dependency with the android-sdk-5.1.1.jar library (we provide it in the release package of the XBee Java Library)

brunoaduarte, 2017-02-06:
As the library was not available in Maven Central before, i manually installed it to my computer local repository. 
I've also included the Android dependency and it worked

>         <dependency>
>             <groupId>com.google.android</groupId>
>             <artifactId>android</artifactId>
>             <version>4.1.1.4</version>
>             <scope>provided</scope>
>         </dependency>

Thanks !

closed 2017-02-06 by brunoaduarte