digidotcom/xbee-java #98: Clean maven dependencies

jvalkeal, 2017-02-09:
When pulling dependency `com.digi.xbee:xbee-java-library:1.2.0` maven complains missing deps for

```
org.rxtx:rxtx:2.2
org.rxtx:rxtx-native:2.2
com.digi:android-sdk:5.1.1
com.digi:android-sdk-addon:3
```

Looks like for `rxtx` latest release is `2.1.7` so I don't know where you got `2.2`. Also you should not define dependencies(use dependency management) in a parent pom as those are probably needed only in samples(well looks like android sample was moved away).

jvalkeal, 2017-02-09:
Right, according to #58 you have probably done a build based on https://github.com/rxtx/rxtx and changed version from `2.2-stabilize-SNAPSHOT` to `2.2`. Looks like `rxtx` has been abandoned which means that you either need to acquire some sort of rights to maintain that project or do your own fork so that you can publish under your maven package id.

Realistically speaking I think it's impossible to use this library in any other library if something would end up to get used in a real app. Companies need to be able to trace down the sources and OSS licenses.

rubenmoral, 2017-02-14:
Hi @jvalkeal,

We haven't made any build of the `rxtx` library, we got the version `2.2pre2` from http://rxtx.qbang.org/wiki/index.php/Download. The source code is also available there. If you don't like that serial library or you want to use another one, feel free to do so (you can even create a pull request).

And regarding the dependencies, they are in the parent pom because they are needed to compile the library too. If you are compiling with maven, you just need to specify the dependency `com.digi.xbee:xbee-java-library:1.2.0`. For Gradle, you also need to add the Digi's Maven repo to the list of repositories:

```
repositories {
    jcenter()
    maven {
        url "http://ftp1.digi.com/support/m-repo"
    }
}
```

Regards.

jvalkeal, 2017-02-14:
U didn't understand what I meant by those deps. You've defined those deps with `dependency` tags and thus all child projects will get all deps even if those are not required. Core parts don't need deps for android. Parent pom should define deps and versions in `dependencyManagement` and then every child project should pick whatever is needed. It's mentioned in [1].

Then if you pull out `xbee-java-library` you are not force fed android libs.

[1] https://maven.apache.org/guides/introduction/introduction-to-dependency-mechanism.html

jvalkeal, 2017-02-14:
Ah, you actually import android libs in a library. Hmm, have you considered making a separate module for android support?

rubenmoral, 2017-02-17:
Thanks for your suggestions, we will try to apply them for the next release.

bertcoenen, 2019-04-02:
What is the timeframe for this enhancement?