digidotcom/xbee-java #105: RXTX serial library crashing on closing port in macOS

keplerian, 2017-04-13:
Case ID: 00278849

We are experiencing problems on macOS when using the latest version of the RXTX serial library with Digi’s XBee Java library:
•	macOS version: 10.11.6
•	XBee Java library version 1.2.0
•	RXTX serial library version 2.2

I’ve attached the error log. Apple identifies the problem as [librxtxSerial.jnilib+0x312f]  Java_gnu_io_RXTXPort_interruptEventLoop+0x6b in the error log.

It happens on closing the serial port with the RXTX serial library.
It seems like it is related to this post: [](http://stackoverflow.com/questions/41785874/java-rxtx-lib-segv-error-on-mac)
I double checked, and I am definitely using the latest version of the RXTX native libraries (version 2.2 pre2).

To prove that it is not my application I reproduced the problem with Digi’s own example code: xbee_java_library/examples/communication/SendDataSample

Here are my XStick driver details as well (from About this MAC > System Report… > Hardware > USB > XStick):

  Product ID:       0x6001
  Vendor ID:        0x0403  (Future Technology Devices International Limited)
  Version:             6.00
  Speed: Up to 12 Mb/sec
  Manufacturer: Digi
  Location ID:      0x14200000 / 6
  Current Available (mA):               1000
  Current Required (mA):              100
  Extra Operating Current (mA): 0

[hs_err_pid7372.log.txt](https://github.com/digidotcom/XBeeJavaLibrary/files/918371/hs_err_pid7372.log.txt)

rubenmoral, 2017-04-18:
Hi @keplerian,

We are investigating your issue. Could you please confirm that you are having this problem when using the Send Data sample as is with just one XBee module (in your case the XStick)?

Thanks.

keplerian, 2017-04-18:
Yes @rubenmoral, I was using just the XStick and the Send Data sample. I think the problem is quite well known. [](https://github.com/NeuronRobotics/nrjavaserial) mentions exactly the problem that I am experiencing in their bug fix list.

rubenmoral, 2017-04-18:
Where is the problem described? Do you mean

> Fixed the memory access error that causes OS X to crash the JVM when serial.close() is called.

We are not able to reproduce it in the same environment using the XStick and the example.

keplerian, 2017-04-18:
Yes, that is it. 

The RXTX website has been down for a while now, so I cannot refer you there. I've attached the jnilib file that we are using. It originated from the rxtx-2.2pre2-bins archive that was downloaded previously from the RXTX website. As far as we could tell, this is the latest available binary file.
[librxtxSerial.zip](https://github.com/digidotcom/XBeeJavaLibrary/files/931313/librxtxSerial.zip)

Possibly related problems:
- http://stackoverflow.com/questions/5979305/java-serialport-close-blocks
- http://marc.info/?l=rxtx&m=123544269723586&w=2
- https://community.oracle.com/thread/1294323

rubenmoral, 2017-04-19:
We can confirm that the JVM crashes when opening the serial connection with that native library. Why aren't you using the library that we provide in the release zip file?

keplerian, 2017-04-20:
We did use the library that you provide in the release zip file, but it was causing another problem. RXTX was blocking after a while, causing us to lose communication with our devices on macOS. I used a few thread dumps to confirm that was the cause. Should I open another issue for this problem?

I decided to try the latest libraries (which I did state in the original report at the top), since all the other native libraries that you provide for the other platforms were identical to the "official" 2.2pre2 libs. So I thought somebody just forgot to update the mac library. Either way, neither library is robust and stable in our system on macOS.

Why are you using another library for macOS than the latest (2.2pre2 previously) available on the RXTX website? Which version is it?

keplerian, 2017-04-23:
By the way: we tried rebuilding the library to work with the [NrJavaSeria](https://github.com/NeuronRobotics/nrjavaserial)l library, since it fixes a few known bugs in RXTX. So far it's a lot more stable on macOS. 

We would still prefer not to use a different serial library than the XBee Java library, but at the moment (because of all the RXTX issues) that is not an option.

rubenmoral, 2017-04-24:
Sorry for the delayed message, @keplerian. We provide RXTX just as an example of serial library, we know that it is not the most robust library but one of the reasons why we chose it was its license. There are others such as SerialIO, jSSC or the one that you mention that can be easily replaced by RXTX in the XBee Java Library. Maybe we have to ask ourselves whether we should use another one.

keplerian, 2017-04-26:
Judging by the number of related issues (https://github.com/digidotcom/XBeeJavaLibrary/issues/39 and https://github.com/digidotcom/XBeeJavaLibrary/issues/99) your users are having with this serial library, I agree that it might be a good idea to ask that question ;) 

NrJavaSerial is probably as close to a drop in replacement as you are going to get for RXTX and it seems like it has the same license. jSSC also uses GPL. Probably the only choice that will cause a licensing problem is SerialIO.

rubenmoral, 2017-04-26:
Yeah, we have already added a task in our backlog for the next release, sorry for the inconveniences...

The idea is to merge #40 and give more options to the user regarding the serial library to use. I mentioned SerialIO because we use it in other products, but we cannot provide it here because of its license (however if a user has purchased it, it would be very easy to integrate it as well).