digidotcom/xbee-java #106: RXTX POM incomplete

rhwood, 2017-05-10:
I get the following errors building a project that depends on this library:
```
Downloading: file:///Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/lib/org/rxtx/rxtx/2.2/rxtx-2.2.pom
Downloading: https://repo.maven.apache.org/maven2/org/rxtx/rxtx/2.2/rxtx-2.2.pom
Downloading: http://ftp1.digi.com/support/m-repo/org/rxtx/rxtx/2.2/rxtx-2.2.pom
[WARNING] The POM for org.rxtx:rxtx:jar:2.2 is missing, no dependency information available
Downloading: file:///Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/lib/org/rxtx/rxtx-native/2.2/rxtx-native-2.2.pom
Downloading: https://repo.maven.apache.org/maven2/org/rxtx/rxtx-native/2.2/rxtx-native-2.2.pom
Downloading: http://ftp1.digi.com/support/m-repo/org/rxtx/rxtx-native/2.2/rxtx-native-2.2.pom
[WARNING] The POM for org.rxtx:rxtx-native:jar:macosx:2.2 is missing, no dependency information available
```
This happens for *every* build, but all other dependencies were downloaded once and reused. Can you please fix the RXTX POM or use an alternate RXTX release such as http://search.maven.org/#artifactdetails%7Corg.bidib.jbidib%7Cjbidibc-rxtx-2.2%7C1.6.0%7Cjar

pontusborg, 2019-03-21:
We are still struggling with this problem, any solutions?

exabrial, 2020-03-17:
Still a problem in 2020