digidotcom/xbee-java #107: Facing issues connecting XBee with Android OS

Swarna1310, 2017-05-15:
Hello I am using SX development board with onboard XBee-Pro SX RF pad module for connecting to the android device and send message. I am facing issues, not able to even open connection. It throws error. Does the library support XBee Pro SX rf module? it is the latest XBee version released.

rubenmoral, 2017-05-22:
Hi @Swarna1310,

Yes, the library supports XBee SX modules. The latest release is v1.2.0.

We have been investigating your problem and it seems that Android is not recognizing properly that type of boards. With the same kit as yours we were able to communicate with the module using the standard XBee Java Library for PC, but we got the same error when tried in Android.

After some debugging, we found out that Android does not recognize the FTDI chip that incorporates that board. However, with other boards such as XBIB-U-SS it works successfully. The problem is that the XBee SX module of your kit is soldered to the board, so you could desolder and use it with the board I mentioned before.

Regards.

Swarna1310, 2017-05-22:
Hi @rubenmoral  ,
Thanks a lot for your response. I have a query related to the Java Library in PC. As android connection was giving errors. I tried communicating with the module using XBee Java library for PC. But faced some error in that. 

I have pasted the code and error I am getting below.  I tried connecting the module in my mac.

public class MainApp {

private static final String PORT = "/dev/tty.usbserial-FTKDZ7DC";

private static final int BAUD_RATE = 9600;
private static final String DATA_TO_SEND = "Hello XBee World!";
public static void main(String[] args) {
XBeeDevice myDevice = new XBeeDevice(PORT, BAUD_RATE);
System.out.println("device info:"+myDevice.toString()+"~~~"+myDevice.isOpen());
byte[] dataToSend = DATA_TO_SEND.getBytes();
try {
myDevice.open();
System.out.format("Sending broadcast data: '%s'", new String
(dataToSend));
myDevice.sendBroadcastData(dataToSend);
System.out.println(" >> Success");
} catch (XBeeException e) {
System.out.println(" >> Error");
e.printStackTrace();
System.exit(1);
} finally {
myDevice.close();
}
}
}

I am getting the following error:( the text below ' >>Error' )

device info:[/dev/tty.usbserial-FTKDZ7DC - 9600/8/N/1/N] ~~~false

WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.2pre1
	native lib Version = RXTX-2.2 (CVS snapshot 2011.02.03, modified by CMU CREATE Lab, http://code.google.com/p/create-lab-commons/)

 >> Error
com.digi.xbee.api.exceptions.XBeeException: Error reading device information.
	at com.digi.xbee.api.XBeeDevice.open(XBeeDevice.java:367)
	at com.digi.xbee.example.MainApp.main(MainApp.java:22)
Caused by: com.digi.xbee.api.exceptions.ATCommandException: There was a problem sending the AT command packet. > Invalid command
	at com.digi.xbee.api.AbstractXBeeDevice.checkATCommandResponseIsValid(AbstractXBeeDevice.java:1980)
	at com.digi.xbee.api.AbstractXBeeDevice.sendParameter(AbstractXBeeDevice.java:2227)
	at com.digi.xbee.api.AbstractXBeeDevice.getParameter(AbstractXBeeDevice.java:2160)
	at com.digi.xbee.api.AbstractXBeeDevice.readDeviceInfo(AbstractXBeeDevice.java:485)
	at com.digi.xbee.api.XBeeDevice.open(XBeeDevice.java:365)
	... 1 more

rubenmoral, 2017-05-29:
Hi @Swarna1310,

I think the problem you are having is related with #101, there was an issue when initializing the SX module in the library. We fixed it in master (see commit 6ca1683250454e364b2bf5830fbeb15935438600).

If you want to use the fix in your application, you can compile and package the library (`mvn clean package`) and then replace the `xbee-java-library-1.2.0.jar` file of your application with the generated artifact.

Regards.