digidotcom/xbee-java #116: RE: THT Grove Development Board and https://github.com/digidotcom/XBeeJavaLibrary/releases/tag/v1.2.1

silver2row, 2017-10-05:
Hello,

My name is Seth. I got a Wireless Connectivity Kit. It has two THT Grove Development Boards and two XBee S1 modules. Currently, I am a starter. I am starting out with NetBeans and software development for these Xbee modules.

I tried: https://github.com/digidotcom/XBeeJavaLibrary/releases/tag/v1.2.1 and the older model of the software v1.1.1.

Now, I followed the instructions to the T. I have tried over and over again. Here is my read out from running the software with my THT Grove Development boards attached and communicating via NetBeans.

    run:
    +-----------------------------------------------+
    |          Receive Digital Data Sample          |
    +-----------------------------------------------+

    java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path thrown while loading 
    gnu.io.RXTXCommDriver
    Exception in thread "main" java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path
     	    at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	    at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	    at java.lang.System.loadLibrary(System.java:1122)
	    at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:123)
	    at com.digi.xbee.api.connection.serial.SerialPortRxTx.open(SerialPortRxTx.java:165)
	    at com.digi.xbee.api.XBeeDevice.open(XBeeDevice.java:334)
	    at com.digi.xdmk.receivedigitaldata.MainApp.main(MainApp.java:54)
    Java Result: 1
    BUILD SUCCESSFUL (total time: 0 seconds)

Although I am reaching a successful build, at least this is what is printed in my terminal, my info. (packets) on the XCTU is not distinguishable. 

Seth

P.S. If you or someone you know are having complications reading an i2c device on DIO1 with this specific development board, please let me know. Please add some feedback if you can. I posted on the Digi Forum like the instructions said but I have not been accepted for posting yet (for some reason).

silver2row, 2017-10-05:
Hello Again,

Another reason I have made this "issue" is that the v1.2.1 download keeps failing.

Seth

P.S. Are other people having issues?

rubenmoral, 2017-10-05:
Hi Seth,

That exception means that you don't have the RXTX library in the Java path. I recommend you to follow the instructions described in the XBee Java Library documentation (https://www.digi.com/resources/documentation/digidocs/90001438/Default.htm#tasks/t_config_project_netbeans.htm) to configure your NetBeans project with all the required dependencies.

What do you mean by "the v1.2.1 download keeps failing"? Aren't you able to download it from https://github.com/digidotcom/XBeeJavaLibrary/releases/download/v1.2.1/XBJL-1.2.1.zip?

Regards,
Ruben

silver2row, 2017-10-05:
Hello,

Do I need to put the path as -Djava.library.path=C:\taco\tacoII\libs\native\Windows\win32 or without the C:\? I also used the cmd prompt on WIn 10 to see my java -version. It did not state 32 or 64. I am running a 64 bit machine. 

Seth

P.S. I am working on a Windows computer for now. Oh and sir, I got the download for v1.2.1 to work finally.

rubenmoral, 2017-10-05:
Hey Seth,

You can use an absolute or relative path, both ways are fine. If you don't know the architecture of your Java Virtual Machine, try launching the example with the 32-bit native path as you have now. If the example throws an exception with the message ``Can't load IA 32-bit .dll on a AMD 64-bit platform``, that means that you have a 64-bit JVM, so you should change the path to ``C:\taco\tacoII\libs\native\Windows\win64``.

Regards,
Ruben

silver2row, 2017-10-05:
Hello, 

Thank you for your feedback. I know I am out of line for asking the PATH question. I am sure you are very busy. I appreciate you clearing that info. up for me.

Seth

P.S. If I come across another issue, I will research the idea first this time.

rubenmoral, 2017-10-06:
Hi Seth,

You're welcome! Don't worry, ask us whatever you want.

Regards,
Ruben

silver2row, 2017-10-06:
Sir,

Thank you for saying so. I actually have an issue. I tried to open a .jar file with Eclipse. I am getting an odd set of error codes in Eclipse when messing with loading .jar files. 

I went to the Eclipse forums to get answers but I do not think there are many people that frequent that legacy forum. Anyway, if you have time, please view my submission: https://www.eclipse.org/forums/index.php/t/1089264/. The second photo in the dialog is whey I typed up is this specific issue. 

I know you may not use Eclipse for running Java Software but I thought it was worth a shot.

Seth

P.S. I tried it on NetBeans too. I think I am going to update NetBeans and see what happens afterward.

rubenmoral, 2017-10-06:
Hi Seth,

Why are you trying to open a .jar file? You don't have to, they are Java libraries which contain compiled code and are needed by the XBee Java Library to work.

I see in the first image that you have already created the Eclipse project and configured the libraries, so the last step is to copy the source code to the ``src`` folder as explained at https://www.digi.com/resources/documentation/digidocs/90001438/Default.htm#tasks/t_add_application_source_code_eclipse.htm.

Regards,
Ruben

silver2row, 2017-10-06:
Hello,

You were right. The software just needed to be added but I received an error. Please view here:

    java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path thrown while loading 
    gnu.io.RXTXCommDriver
    Exception in thread "main" java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path
	    at java.lang.ClassLoader.loadLibrary(Unknown Source)
	    at java.lang.Runtime.loadLibrary0(Unknown Source)
    	    at java.lang.System.loadLibrary(Unknown Source)
	    at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:123)
	    at com.digi.xbee.api.connection.serial.SerialPortRxTx.open(SerialPortRxTx.java:165)
	    at com.digi.xbee.api.XBeeDevice.open(XBeeDevice.java:337)
	    at com.digi.xbee.example.MainApp.main(MainApp.java:22)

Is this because I placed the rxtxSerial .jar file in the wrong path? 

Seth

rubenmoral, 2017-10-06:
Hi Seth,

That's the same error that you reported in your first message. Did you add the RXTX library to the path? Here is explained how to do it in Eclipse: https://www.digi.com/resources/documentation/digidocs/90001438/Default.htm#tasks/t_config_project_eclipse.htm.

Regards,
Ruben

silver2row, 2017-10-06:
Yes Sir,

They changed the IDE some since this was produced, I guess. I had to look around on how to produce the correct PATH. I got the PATH put in and ready. I get an I/O Exception Error.

See here:

    C:\Program Files\Java\jre1.8.0_141\lib\gnu.io.rxtx.SerialPorts
    IOException!

I looked on my screen and this is what showed up before the above print out on the terminal:

[image]

I know this is testing your patience. If you know of this idea, please reply. I will be patient.

Seth

P.S. In the meantime, I will go to the legacy forum on Eclipse and post there.

silver2row, 2017-10-08:
Hello Sir,

I just updated NetBeans. I will attempt this issue another time and get back to you. I was using Netbeans 8.0.2 and just updated to 8.2.0. I will let you know how things go.

Seth

P.S. If there are specific docs. stating specific error codes for Java, please let me know something. I am new to Java and need some support at times. I am really sorry for flooding this post so many times. Please forgive me.

silver2row, 2017-10-09:
Hello Sir,

Sorry for all these posts. I just was told that the Grove Development boards I own, the THT type, do not have processors or chips. 

Seth

P.S. I was plugging these boards in with the XBee attached to a computer. I have a Linux machine called the BeagleBone Black. I will plug one of the Grove boards into this machine instead to have a processor to work with.

closed 2017-10-09 by silver2row