digidotcom/xbee-java #120: WARNING:  RXTX Version mismatch

HeroErix, 2018-03-19:
I have followed the instructions for Getting started with XBee Java Library here: https://www.digi.com/resources/documentation/digidocs/90001438/Default.htm

When attempting to run the example program in eclipse I get the following output.
[image]
And the receiver will receive what is shown no matter what message I try to send.
[image]

I have previously used these two devices to communicate, the sender connected to XCTU and the receiver connected to an arduino, so the settings on each device should be correct. Any help in solving these errors would be appreciated.

hgonzaleDigi, 2018-03-20:
It seems that your local device is working in AT mode. This library only supports "API" and "API with escapes" operating modes.
So, in order to use this library, verify the AP setting of your local device reports values of 1 or 2.

HeroErix, 2018-03-20:
Thank you. I changed both devices to API mode and I can now send using the example application.

closed 2018-03-21 by hgonzaleDigi