digidotcom/xbee-java #121: Setting NI parameter bug?

aang7, 2018-03-20:
First of all when I use either the RemoteXBeeDevice class or RemoteATCommandPacket class, for example,  if I set the paremater value to "BB". When I check the new value on the remote, it has the value of "B0B". So the set value is different. 

Also I don't know how to "WRITE" the changes after changing that parameter ("the Network ID"). I know it's because the remote it's already on the new set network, but I need a way to send also the "write changes" command at the same time.

This is how I am sending the instruction:
`remote.setPANID(new byte[]{(byte)0xB, (byte)0xB});`
or
`RemoteATCommandPacket packet = new RemoteATCommandPacket(1, remote.get64BitAddress(), remote.get16BitAddress(), XBeeTransmitOptions., "ID", new byte[]{(byte)0xB, (byte)0xB});`

hgonzaleDigi, 2018-03-21:
Hello @aang7 

There is no bug with this feature, so here is how it works.

If you want to change the ID command of a remote device, you must send all the commands you want before applying them. This means that the remote device will get the commands but not apply them until you decide so.

To disable your device from applying changes, use this method
`remote.enableApplyConfigurationChanges(false);`

For your case, your first approach was the correct one, the only error there is that you need to send only one byte (0xBB) and not 2 separate bytes (0x0B and 0x0B). That was the reason you saw 0xB0B.

This means that your code to set the Pan ID and write the change should be something like this:
```
remote.enableApplyConfigurationChanges(false);
remote.setPANID(new byte[]{(byte)0xBB});
remote.writeChanges();
remote.enableApplyConfigurationChanges(true);
```

aang7, 2018-03-26:
@hgonzaleDigi Thank you very much!
Sorry for replying until now. I have another question for you. What exaclty does the XBee do when we call to _applyChanges_ method? I mean, it sends one packet or command  for all the previous calls between _enableApplyConfigurationChanges_ in one transmission or one by one? 
Thank you again!

hgonzaleDigi, 2018-03-26:
Well, by default, the library assumes that every remote AT command should be applied.

This means, that each command is sent to the remote device and, through some configurations, our local device tells the remote one that an AC (Apply changes) command should be issued.

Applying a command means that the node starts using the value of that command internally. You can transmit a remote command without applying it, for example the ID command, but the device will remain in the network until the AC command is sent. This leaves you with the opportunity to send more commands, and once you are done, you can send the AC command and all of them will be applied at the same time.

As the library is written, after each remote command and AC is sent, which means that after you send the ID command, the change will be applied and automatically, your node will exit the network, which would then cause you not to be able to talk to it anymore.

With the above code, you disable the automatic application of commands, until you are done sending them, this way you can issue the WR (write) command without any errors. I did not issue the

To answer your question, the applyChanges sends an AC remote AT command.
The enableApplyConfigurationChanges method only tells the module to apply the changes of each command that is sent (send an AC after each remote AT command).

In your case you don't need to send an AC manually, because the WR command (writeChanges method) does that implicitly.

aang7, 2018-03-26:
Many things make sense right now! Really, Thank you so much! :) I'm gonna mark this as closed.

closed 2018-03-26 by aang7