digidotcom/xbee-java #123: Library is now crippled with Android dependencies

bhaecke, 2018-04-09:
Any reason why core classes are crippled with android dependencies that have absolutely nothing to do with XBee's?

alcarraz, 2019-04-25:
same problem here, I had to strip away all android sdk references in order to build against a standard jdk, because android sdk redefines lots of core apis without standard methods like stream() in collections

alcarraz, 2019-05-03:
Hi @bhaecke, just in case you still didn't, or for others that may reach this issue by searching, I've made a fork  (https://github.com/alcarraz/XBeeJavaLibrary) and stripped of all the android dependencies. 

I hope this issue is dealt with before next release, which I would like to be sooner rather than later since the last one does not support Xbee 3.

I think that android support dhould be optional and can be implemented as a second library with device implementations that extend the base ones and add that support.

bhaecke, 2019-05-03:
Hi @alcarraz I got rid of all the Android dependencies as well.