digidotcom/xbee-java #124: Receive Timeout doubt

aang7, 2018-04-12:
Hi, I have a doubt when working with the receive timeout parameter. I'm trying to get a parameter from a remote XBee using Digimesh protocol, with the XBee PRO model. Before I request the parameter value, I set the receive timeout of my local XBee device as this:
`myDevice.setReceiveTimeout(300000);`

Then I request the value:
`remoteXBeeDevice.getDIOValue(IOLine.DIO11_PWM1);`

But I receive an exception:
> There was a problem sending the AT command packet. > TX failure

Why I'm getting this error too fast if I have a longer receive timeout ?

_TX failure_ It's an error of trying to send the command, but why if with others remotes the same code works?

rubenmoral, 2018-04-13:
Hi @aang7,

What type of modules are you using? Could you please send us the hardware and firmware versions (HV and VR)?

Also, are you able to read any other parameter from that remote module (PAN ID or other IO line for example), or you always get that error?

Thanks.

aang7, 2018-04-13:
I'm using
**Firmware version:** 9000
**Hardware version:** 2D42
**DigiMesh 2.4  TH PRO**

And so far, I'm not able to read another parameter.

rubenmoral, 2018-04-13:
Thanks @aang7,

Are you sure the two XBee modules are in the same network and can communicate each other? Can you verify with XCTU that you can discover the remote module from the local one?

How do you obtain the `remoteXBeeDevice` object?

Thanks.

aang7, 2018-04-13:
> Are you sure the two XBee modules are in the same network and can communicate each other? 

I'm sure that the modules are in the same network.

> Can you verify with XCTU that you can discover the remote module from the local one?

The XCTU can discover the remote module from the local one. But after a few re-scanning.

> How do you obtain the remoteXBeeDevice object?

Instantiating the `remoteXBeeDevice` by the MAC address.
As this:
`RemoteXBeeDevice remoteXBeeDevice = new RemoteXBeeDevice(myDevice, new XBee64BitAddress("0013A200410A24FB"));`

rubenmoral, 2018-04-16:
Hey @aang7,

If XCTU takes several scans to discover that remote device, it could be because the communication with it is very poor. A `TX failure` error means that the packet could not reach the remote device, either because it does not exist on the network or it is too far so the packet is lost.

With the information you have provided, we are not able to reproduce the problem. Could you please give us more information about your network (number of modules, distance between them...)?

Thanks.