digidotcom/xbee-java #128: Digi Xbee communication error

Antonyab, 2018-10-05:
Hello,

I am using the DIGI XBEE CELLULAR 3G GLOBAL Development Kit.The following
details are mentioned on the digiXbee module :
DIGI XBee Cellular 3G
XBC-M5-UT-101 B
357520072985670
S# 08J8BAP87NQ4
Contains FCC ID : XPY1CGM6NNN
IC: 8595A-1CGM5NNN

Can someone please help me to know if my module is programmable or not ?
In fact, I am having the same issue mentionned in this link :
https://github.com/digidotcom/XBeeJavaLibrary/issues/57
Summary of the issue :
com.digi.xbee.api.exceptions.InvalidOperatingModeException: Could not
determine operating mode.

I need to know what to do in order the solve this problem.

Thanks in advance.

Regards,
Antony Abi Rached (Kalima Systems)

brandonmoser, 2018-10-05:
@Antonyab what do you mean by programmable? Yes, the module is programmable but is not currently supported by this library, as noted [here].(https://github.com/digidotcom/XBeeJavaLibrary/blob/master/release_notes.txt) You'll need to use our XCTU application, https://www.digi.com/products/xbee-rf-solutions/xctu-software/xctu.

Antonyab, 2018-10-05:
Thanks for the reply.
In fact i am facing the same issue that is mentionned in the following link
https://github.com/digidotcom/XBeeJavaLibrary/issues/57.

When i use a java api example with the digi xbee mounted on the demo board
given in the kit, i can communicate normally with the module.
When I connect the module directly to the uart of the raspberry pi, the
same java api example is not working. I get the following error:
com.digi.xbee.api.exceptions.InvalidOperatingModeException:
Could not
determine operating mode.

do you know the cause of this problem?
Thanks for the help.

Regards,
Antony

Antonyab, 2018-10-09:
Hello again @brandonmoser,

I will explain for you the tests that I did with the module that I mentioned in my first post.

* First of all I downloaded XCTU application and I tested to send a message to an echo server. The example is mentioned in this documentation : https://www.digi.com/resources/documentation/digidocs/90001541/default.htm#tasks/t_echo_server.htm%3FTocPath%3DConnection%2520examples%7C_____1

* After that, I configured the module in API Mode (API Mode Without Escapes) in order to use the java API with the module.  

* I downloaded XBeeJavaLibrary from this link: https://github.com/digidotcom/XBeeJavaLibrary/releases
I took the .jar and test it the same example (send message to echo server). I got the same result. It worked !!! 
P.S : the module is mounted on the demo board and is connected to my pc using USB. The OS of my pc is windows.

* I tried to execute the same example on a raspberry pi. So I generated an executable and runned it on raspberry but it didn't work due to a serial port issue (RXTXlibrary). 

* So I decided to use purejavacomm instead of RXTXlibrary to handle serial port using Java. So I downloaded the source code of the XbeeJavaLibrary and made the changes to implement purejavacomm instead of RXTX library.

* After the changes that I made, I tested again on the raspberry pi and it worked perfectly. P.S : the module is mounted on the demo board and is connected to the raspberry pi using USB.

* Now I want to try using the module directly on the UART of the rapsberry pi without being mounted on the demo board. So I connected the module to UART and I tried the same example that I tested before. Here is the problem. When I execute the example, I got the following error :  

com.digi.xbee.api.exceptions.InvalidOperatingModeException: Could not determine operating mode.

I dont know the cause of this problem. The following issue https://github.com/digidotcom/XBeeJavaLibrary/issues/57 describe the same problem that I have. But after contacting the digi support, I got the following response:
*********************************************************************************************************
I looked through that GitHub issue. The old Programmable XBee modules had a Freescale C processor that sat between the UART and the SOC, and that what was causing the issue reported on GitHub. The C processor had no code on it by default, and blocked communication between the UART and the SOC until it was explicitly put into serial bypass mode or loaded with some other application code.
The new Xbee Cellular and XBee3 products, like the one that you have, feature embedded MicroPython which is integrated into the SOC and is not accessed unless AP is explicitly set to 4. So yes, the Xbee Cellular you have is programmable, but that is not what is causing your issue. 
If you have AP=1 for API mode, and you're still receiving that error, then you will want to file an issue on the GitHub repository. The XBee Libraries were designed for our previous ISM XBee products, and might not be updated for XBee Cellular products.
*********************************************************************************************************

does someone know the cause of the problem that I am facing ?

Thanks in advance for your help.

Regards,
Antony

brandonmoser, 2018-10-09:
@Antonyab what operating mode are you running. The supported modes are 0-3, as found [here](https://github.com/digidotcom/XBeeJavaLibrary/blob/master/library/src/main/java/com/digi/xbee/api/models/OperatingMode.java#L25-L29).

Antonyab, 2018-10-09:
@brandonmoser I am using API mode. the value of the variable AP is 1.
In fact I configured this value using the XCTU application.
I should also specify the mode in my java application?

Regards,
Antony

brandonmoser, 2018-10-09:
@Antonyab I think you should output your operating mode from the Java library to confirm it is seeing your AP value of 1. There may be an inconsistency or issue there, since the XBee Cell module is not currently supported.

brandonmoser, 2018-10-09:
@rubenmoral do you have any suggestions?

Antonyab, 2018-10-10:
@brandonmoser When I configure the variable AP using XCTU application to API Mode (API = 1), the configuration will be saved on the module or on the board? 
If on the board, that would explain my problem. So what do you think about that?

Regards,
Antony

brandonmoser, 2018-10-10:
All values are saved to the module, the dev boards do not have memory.