digidotcom/xbee-java #130: Getting InterfaceNotOpenException while sending data

SaiCharitha75, 2019-01-11:
Hi Digi,

I am using xbee-java-library-1.2.1, Jar version = RXTX-2.2pre1, native lib Version = RXTX-2.2pre2
I need some clarifications regarding com.digi.xbee.api.exceptions.InterfaceNotOpenException: The connection interface is not open.

I have attached coordinator to Raspberry pi USB port and in my program I have instantiated an xbee device object which is  global static object and I have opened the communication interface. I am checking whether communication interface is open or not before sending data to xbee device. 

Up to some hours Communication interface is opened and I am able send data to xbee device but at some point I am getting “InterfaceNotOpenException”.

(Note: I am not closing the instantiated xbee device object)

Can you please let me know what are  the possibilities of getting this exception?

 
```
static XBeeDevice myDevice ;

boolean serialcheck()
	  {
	    try
	    {
	         PORT= Serial.list()[((int)this.config.Serialport)];
	         myDevice = new XBeeDevice(PORT, 9600);
	         myDevice.open();
	         myDevice.addDataListener(new MyDataReceiveListener());
	         
	         return true;
	 
	    }
		return false;}
		
Void sendData()
	{
           if(myDevice.isOpen()){
			RemoteXBeeDevice remoteDevice = new RemoteXBeeDevice(myDevice,
						    new XBee64BitAddress(address));
			myDevice.sendDataAsync(remoteDevice, data.getBytes());
					
				}
}
```

rubenmoral, 2019-01-14:
Hi @SaiCharitha75,

That's really weird, if you are getting that exception it means that, for some reason, the connection interface with the device has been closed.

Could you please send us the stack trace of the exception? It would be very helpful in order to understand what is happening.

SaiCharitha75, 2019-01-21:
Hi @rubenmoral,

Please find  the Stack trace of the exception 

MethodName: RequestNodes_Missed   lineno:1229   info:Error   Threadname:ForkJoinPool.commonPool-worker-2   **description:**The connection interface is not open. **Cause:** null
 **StackTrace:** com.digi.xbee.api.exceptions.InterfaceNotOpenException: The connection interface is not open.
	at com.digi.xbee.api.AbstractXBeeDevice.sendXBeePacket(AbstractXBeeDevice.java:1229)
	at com.digi.xbee.api.AbstractXBeeDevice.sendXBeePacketAsync(AbstractXBeeDevice.java:1195)
	at com.digi.xbee.api.AbstractXBeeDevice.sendAndCheckXBeePacket(AbstractXBeeDevice.java:1502)
	at com.digi.xbee.api.XBeeDevice.sendDataAsync(XBeeDevice.java:750)
	at com.digi.xbee.api.XBeeDevice.sendDataAsync(XBeeDevice.java:785)
	at digimeshGateway.ReceiveAndMessageIot.RequestNodes_Missed(ReceiveAndMessageIot.java:1056)
	at digimeshGateway.ReceiveAndMessageIot$MessageSender$1.run(ReceiveAndMessageIot.java:993)
	at java.util.concurrent.CompletableFuture$AsyncRun.run(CompletableFuture.java:1626)
	at java.util.concurrent.CompletableFuture$AsyncRun.exec(CompletableFuture.java:1618)
	at java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:289)
	at java.util.concurrent.ForkJoinPool$WorkQueue.runTask(ForkJoinPool.java:1056)
	at java.util.concurrent.ForkJoinPool.runWorker(ForkJoinPool.java:1692)
	at java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:157)

SaiCharitha75, 2019-01-21:
Hi @rubenmoral 

I am also getting java.lang.IllegalArgumentException sometimes while sending data to remote xbees.
As we are not generating any frameid in our code, can you please explain the scenario for getting this issue.

**Code:**
```
try{
					if(myDevice.isOpen()){
						
					RemoteXBeeDevice remoteDevice = new RemoteXBeeDevice(myDevice,
						    new XBee64BitAddress(address));
					myDevice.sendDataAsync(remoteDevice, data.getBytes());
					
						}
					}
					catch(Exception e){
						StringWriter sw = new StringWriter();
						e.printStackTrace(new PrintWriter(sw));
						println("MethodName:sendDataAsync()of MessageSenderClass   lineno:" + e.getStackTrace()[0].getLineNumber() + "   " + "info:" + "Error" + "   " + "Threadname:" + Thread.currentThread().getName() + "   " + "description:" + e.getMessage()+" "+"Cause: "+e.getCause()+" "+"StackTrace:"+sw.toString());
					}

```
**Exception:**
MethodName:sendDataAsync()of MessageSenderClass   lineno:188   info:Error   Threadname:pool-8-thread-4   description:Frame ID must be between 0 and 255. Cause: null StackTrace:java.lang.IllegalArgumentException: Frame ID must be between 0 and 255.
	at com.digi.xbee.api.packet.common.TransmitPacket.<init>(TransmitPacket.java:188)
	at com.digi.xbee.api.XBeeDevice.sendDataAsync(XBeeDevice.java:749)
	at com.digi.xbee.api.XBeeDevice.sendDataAsync(XBeeDevice.java:785)
	at digimeshGateway.ReceiveAndMessageIot$MessageSender.run(ReceiveAndMessageIot.java:967)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

rubenmoral, 2019-01-21:
Hi @SaiCharitha75,

What type of XBee modules are you using? We need the hardware (**HV**) and firmware (**VR**) versions in order to try to reproduce your issue.

I understand by your first message that the code is running on the RPi, the local device is the coordinator and you are sending data to a remote device from that one, right?

SaiCharitha75, 2019-01-21:
Hi @rubenmoral 

Yes, I am running my code on the RPI with localdevice (coordinator) to send data to remote xbee module.
As we are using two types of xbee modules , i am providing HV and VR of both modules.

**Module: XBee-PRO 900HP (S3B) DigiMesh.
Specs: 900MHz, 250mW, RPSMA, 200Kbps
Protocol:Digimesh
PartNo-XBP9B-DMST-002
Product family:XBP9B-DM
FunctionSet: XBee PRO 900HP 200k
Firmware Version(VR): 8075
HV-234A

Module: DIGI XBEE-PRO
Specs: 2.4 Ghz,63mW,RPSMA,250kbps
protocol: zigbee
Part No: xbp24cz7sit-004
VR-4060
HV-2D46**

aylashiv, 2019-01-24:
Yes I did face the same issue, with my XBP9B-DMST002 version of Sub 1 GHz module. Interfaced to one of the  USB port using a FTDI board, Raspberry Pi 3 Model B.

rubenmoral, 2019-01-28:
Hey @SaiCharitha75, I think we have fixed the issue of the `IllegalArgumentException` when sending data, but are not able to reproduce the one related to the interface not open. I have a small program running on a Raspberry Pi that is constantly sending data between two XBee S3B modules, and it has been working fine so far.

How often are you sending data? Does the coordinator receive any response from the remote one? What is the distance between both modules? If you could share the code with us, that would help a lot (ruben.moral@digi.com).

@aylashiv, what is the issue in your case? Please give us more details so we can investigate it deeper.

SaiCharitha75, 2019-01-29:
HI @rubenmoral ,I am sending the data to remote xbee devices using multiple threads.
For every 1 minute, i am parallely sending the data to  remote xbee devices based on the address of the nodes and through Data Receiver listener , i am receiving the response . 
If any of the node was not responding within the 15 seconds after the request was mode , i am sending another request to particular node based on the address.
Suddenly  after few hours i am getting error messages as " interface not open exception".

SaiCharitha75, 2019-01-29:
Note: I have now where used synchronised method for sending in my code.

rubenmoral, 2019-01-29:
@SaiCharitha75, by what you say, my guess is that the exception could be caused by the data packet which is received by the coordinator. If something is wrong there, then it makes sense that the library closes the connection and you cannot send more messages to the remote devices.

What type of packet are the remote devices sending to the coordinator? How are you handling the receive data in the code?