digidotcom/xbee-java #132: XBee 3 Compatibility

nenstrom, 2019-01-29:
Is there any plans for the XB3 modules to be supported by this library?

Was hoping since the AT commands remain generally the same that I'd be able to use the libraries here.

Using Module XB3-C-A1-UT-001.

Receiving error:

Exception in thread "main" com.digi.xbee.api.exceptions.XBeeException: Error reading device information: Your module seems to be Unknown and NOT Cellular. Check if you are using the appropriate device class.

I've tried AP mode 0, 1, and 2. In mode 0, as expected, I get an error saying I am in AT mode. The other two modes throw the error above.

Is there a work around to still be able to use these libraries or will I have to wait for a future release?

Thanks for any insight you may provide.

alcarraz, 2019-05-03:
Hi, the support for that devices is already in the current master branch, but it has not been released yet.

I had to fork, remove the dependencies of the android sdk (*) and build the library myself.

(*) otherwise my project wouldn't compile because we make heavy use of stream in collections and the android sdk overrides the jdk and does not provide such methods

alcarraz, 2019-10-16:
I think this issue can be closed now

closed 2019-10-17 by rubenmoral