digidotcom/xbee-java #133: Reading Device Info and channels

aang7, 2019-04-16:
Hi there! 
I have found some weird behavior. Let me explain.
I'm working with a network which has different channels but the same ID.
So I have a method that calls _readDeviceInfo()_ to get the essential info about some XBeeRemote. When I call that method I just check if my Local XBeeDevice has the same channel of the requested XBeeRemote, and if it haven't then I just change the Local XBeeDevice channel to be able to communicate with the XBeeRemote. When I try the first time(Calling readDeviceInfo method) it works but in future calls it doesn't, I don't know why. 
The method throws an exception, which is basically the following:
`com.digi.xbee.api.exceptions.ATCommandException: There was a problem sending the AT command packet. > TX failure
	at com.digi.xbee.api.AbstractXBeeDevice.checkATCommandResponseIsValid(AbstractXBeeDevice.java:1980)
	at com.digi.xbee.api.AbstractXBeeDevice.sendParameter(AbstractXBeeDevice.java:2227)
	at com.digi.xbee.api.AbstractXBeeDevice.getParameter(AbstractXBeeDevice.java:2160)
	at com.digi.xbee.api.AbstractXBeeDevice.readDeviceInfo(AbstractXBeeDevice.java:451)
	at gui.NodeInfoController.refreshInfo(NodeInfoController.java:437)
	at gui.NodeInfoController.lambda$initialize$6(NodeInfoController.java:168)`

The exception refers to following line of code:
`            XBeeRemote.readDeviceInfo();
`

Hope someone helps.

rubenmoral, 2019-04-17:
Hi @aang7,

Could you please specify the type of XBee modules you are using (hardware version and firmware version)? This could help us to understand why you are getting that exception.

Also, please make sure you have the latest version of the library (1.2.1). Taking a look to the exception trace, it seems you have the previous version (1.2.0).

Thanks,
Ruben

aang7, 2019-04-17:
**Firmware version**: 9000
**Hardware version**: 2E43
**Product family**: XB24C
**Function set**: DigiMesh 2.4 TH

And yes, I'm using the previous version (1.2.0).
Let me try with the latest one.

aang7, 2019-04-17:
After update to the latest version of the library(v1.2.1). The behavior seems to be same, It throws the same exception.