digidotcom/xbee-java #134: what's up with rxtx and rxtx-native

grimace, 2019-05-24:
seems like this might be a problem using this Library:

```
[WARNING] The POM for org.rxtx:rxtx:jar:2.2 is missing, no dependency information available
[WARNING] The POM for org.rxtx:rxtx-native:jar:macosx:2.2 is missing, no dependency information available
[WARNING] The POM for com.digi:android-sdk:jar:5.1.1 is missing, no dependency information available
[WARNING] The POM for com.digi:android-sdk-addon:jar:3 is missing, no dependency information available
```

Has there been any progress on this?

liuxa86, 2019-07-22:
It looks like that RXTX Version 2.2 hab been removed from maven repo.

        - https://repo.maven.apache.org/maven2/org/rxtx/rxtx/2.2/rxtx-2.2.pom
       - https://repo.maven.apache.org/maven2/org/rxtx/rxtx/2.2/rxtx-2.2.jar
       - https://jitpack.io/org/rxtx/rxtx/2.2/rxtx-2.2.pom
       - https://jitpack.io/org/rxtx/rxtx/2.2/rxtx-2.2.jar

are dead.

bootsy52, 2019-12-03:
What about switching to https://fazecast.github.io/jSerialComm/ for example seems to be more maintained. Ore decapsulate the serial communication so one could use it's own serial library

christianitus03, 2021-01-22:
> What about switching to https://fazecast.github.io/jSerialComm/ for example seems to be more maintained or decapsulate the serial communication so one could use it's own serial library

That's an excellent idea considering how old RXTX is, but wouldn't that require a pretty major rewrite of this entire library?