digidotcom/xbee-java #161: xbee-java crashing with newer java versions

msoldin, 2022-02-04:
*Why*
xbee-java is crashing with newer java systems like JDK11. As far as I know, this only affects 64 bit versions of Java. As far as I know, this only affects 64 bit versions of Java, as the rxtx library is simply too old. I know the compatibility only says Java 8, but Java 8 is no longer supported with updates. I find it an absurdity to expose software to such security risks for customers.

*What*
Replace the rxtx-Library with newer code libraries or write an own wrapper for it.

bramvdpol, 2022-06-13:
Since there was a recent release of this library apparently development is still done on it; Please make this a priority! The only way to use the xbee-java library on recent windows version is by using an ancient JRE8 version - JRE251 from 2020 or older. 

This is not acceptable for production use!

Related: https://stackoverflow.com/questions/63157940/serial-port-reading-error-exception-access-violation-c-rxtxserial-dll0x5

msoldin, 2022-06-13:
@bramvdpol you can replace the serial library by sacrificing the unit tests. I already did that for my master thesis, seems to work fine.

https://github.com/msoldin/xbee-java-nrjavaserial

bramvdpol, 2022-06-13:
All right, thanks for your insight! That does provide some workarounds for the time being. I think I prefer untested code that at least runs on the latest JRE compared to the current situation. I will also look into OpenJDK, that might be the best option for now.
Thanks again!

HomeOfTheWizard, 2026-03-29:
Is there a work planned to resolve this issue ? or should we just consider this library dead ?