digidotcom/xbee-java #167: Uncaught exception: java.lang.IllegalArgumentException: 64-bit address cannot be unknown.

FilipeLimao99, 2023-03-29:
Hello, 
I'm currently using version 1.1.0 of the xbee-java-library.
I'm getting _Uncaught exception: java.lang.IllegalArgumentException: 64-bit address cannot be unknown._ message after broadcasting to a network of 21+ devices. If there are 25 Xbee nodes in the network I get the error, if I have 21 nodes this message is not visible.

After first broadcast:
Uncaught exception: java.lang.IllegalArgumentException: 64-bit address cannot be unknown.
com.digi.xbee.api.XBeeNetwork.getDevice(XBeeNetwork.java:484)
com.digi.xbee.api.connection.DataReader.getRemoteXBeeDeviceFromPacket(DataReader.java:652)
com.digi.xbee.api.connection.DataReader.packetReceived(DataReader.java:551)
com.digi.xbee.api.connection.DataReader.run(DataReader.java:473)
java.lang.Thread.run(), bci=5

After second broadcast 
com.digi.xbee.api.exceptions.InterfaceNotOpenException: The connection interface is not open.
com.digi.xbee.api.AbstractXBeeDevice.sendXBeePacket(AbstractXBeeDevice.java:1439)
com.digi.xbee.api.AbstractXBeeDevice.sendAndCheckXBeePacket(AbstractXBeeDevice.java:1672)
com.digi.xbee.api.XBeeDevice.sendExplicitData(XBeeDevice.java:1340)
com.digi.xbee.api.XBeeDevice.sendBroadcastExplicitData(XBeeDevice.java:1564)
com.digi.xbee.api.ZigBeeDevice.sendBroadcastExplicitData(ZigBeeDevice.java:327)
com.owlet.xbee.OwlXbee.broadcast(OwlXbee.java:371)
com.owlet.xbee.OwlXbee.broadcast(OwlXbee.java:381)
com.owlet.xbee.XbeeSensorBroadcast.broadCast(XbeeSensorBroadcast.java:102)
com.owlet.xbee.XbeeSensorBroadcast$1.run(XbeeSensorBroadcast.java:77)
com.owlet.util.TaskPeriodic$1.run(Task.java:175)
java.util.TimerThread.mainLoop(), bci=244
java.util.TimerThread.run(), bci=1

Can you give me some clarification on why this is happening or if it's fixed in newer versions of the library?

Thanks you,
Filipe Limão

tatianaleon, 2023-03-29:
Hi @FilipeLimao99,

Use the last version of the library and check if this error is happening. If it persists, please provide a way to reproduce the issue: source code, XBee type, firmware version, and any other relevant value.

Best Regards,
Tatiana

FilipeLimao99, 2023-04-19:
Hello @tatianaleon, we are having some issues on upgrading the library and compiling with our code. Our main project is Java ME, and we wanted to make sure that the issue is fixed before we start adapting your library to fit our needs. We estimate a lot of effort doing that and so we wanted to assess the risks before doing so. 
On that note we would be much appreciated if you could give us any leads on the issue being fixed or not. 
Do you perform any stress tests using your library? Or do you know anyone how might have been working with your library and a large number of nodes and traffic?

tatianaleon, 2023-04-19:
Hi @FilipeLimao99,

As far as I know, there is not such problem, but it is difficult to say if something has been fixed when we do not know what to check:

* As I said we would need a way to reproduce the problem, that includes a piece of code, the setup, firmware versions, and the required sequence to see the issue.
* Version 1.1.0 is quite old, preferably we would need a way to reproduce the issue with the latest version.
* The code you are using does not seem to be plain version 1.1.0, it seems it is modified.
Source code lines in your exception stack traces does not match with code lines in version 1.1.0. For example [line 484 of file XBeeNetwork.java](https://github.com/digidotcom/xbee-java/blob/v1.1.0/library/src/main/java/com/digi/xbee/api/XBeeNetwork.java#L484) is a comment line, the same happens with [line 1439 of file AbstractXBeeDevice.java](https://github.com/digidotcom/xbee-java/blob/master/library/src/main/java/com/digi/xbee/api/AbstractXBeeDevice.java#L1439).
This means it is difficult to know if the error is in the original code or in any modifications you did.

My recommendation is to reproduce the issue with version 1.1.0 using a PC for example, that does not imply any kind of adaptation. And when you have it, do the same with the latest version. If the error persist, please let us know and we will investigate it.

Best Regards,
Tatiana